pvvx/ATC_MiThermometer #1: OTA update

michapr, 2020-12-21:
I have used your repo again ;) -  complete sync from TODAY in the morning.
Have you tried the OTA firmware update? I have some problems here (using your flasher page)
At first flashed from original FW to your current release - all fine.
Then have changed the frequency to 16MHz, tried to update via OTA.

...
12:14:13: Detected custom Firmware
12:14:15: Start DFU
12:14:21: Update error: NotSupportedError: GATT operation failed for unknown reason.

After beginning with upload upload is stopped.
After that device is "dead".
I must use serial update (python) with new compiled FW, then is all again ok. Tried two times, same result.
Any idea? ;)

pvvx, 2020-12-21:
check header 'ATC_Thermometer.bin':
[image]
and 
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/cstartup_825x.S#L63

If the head doesn't match, then OTA doesn't work.
A few days ago there was an invalid cstartup_825x.S file.

pvvx, 2020-12-21:
> After that device is "dead".

The OTA timeout has also been changed to the increased 'connect interval':
bls_ota_setTimeout(45 * 1000000); // set OTA timeout  45 seconds
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/ble.c#L47

michapr, 2020-12-21:
All seems fine.
[image]

With "dead" I mean - that even the BLE advertising is not working more at all.

With your devices it is working?

BTW: is your firmware compatible to the old OTA flasher (from ATC repo) - of course without your new options...?

pvvx, 2020-12-21:
Checked more than 3 times. 16 and 24 MHz options. All ok.
```
Log:
23:50:45: Searching for devices
23:50:59: Connecting to: ATC_BB240C
23:51:03: File was selected, size: 45392 bytes
23:51:03: Count: 2837
23:51:04: Detected custom Firmware
23:51:07: Start DFU
23:51:30: Update done after 22.729 seconds
23:51:38: Disconnected.
```

pvvx, 2020-12-21:
> BTW: is your firmware compatible to the old OTA flasher (from ATC repo) - of course without your new options...?

OTA flasher Telink (standatr libs).

michapr, 2020-12-22:
It seems like the python flasher did not erase the flash full.
Could it be possible to check the flash after erase? Is there any such possibility?

I had same effect after flashing a second device - but now is all ok again, after multiple flashing.
As sample at second device have seen that after a flash only (inserted a sys.exit(1) after flash) and reboot the device was still reporting values.... so flash was not successful...

Can you check the "check erased flash" possibility?

mkullber, 2020-12-23:
Ota timeout is too short for uploading original firmware or firmware binary built by atc1441

pvvx, 2020-12-23:
atc1441 uses reset counter-timer:
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/ble.c#L86
```
extern u32 blt_ota_start_tick;
int otaWritePre(void * p) {
	blt_ota_start_tick = clock_time()|1;
	otaWrite(p);
	return 0;
}
```

mkullber, 2020-12-23:
45 seconds is just enough to 50 kb binary, but now I can't revert to original firmware or firmware by atc1441 as they are bigger

pvvx, 2020-12-23:
@mkullber  Sorry, I pressed the wrong key and edited your message.

```
Update done after 22.729 seconds
```
45 seconds - 100 kb

michapr, 2020-12-23:
>Update done after 22.729 seconds

This really depends on the BLE adapter / connection.
I have very big difference between Win PC / tablet / Linux PC - for ATC update between 40 seconds and 1:30 about ....

mkullber, 2020-12-23:
On my phone 44 kb takes just below 40 sec

pvvx, 2020-12-26:
In the next version, the time counter is reset when new OTA data is received.

closed 2020-12-27 by pvvx