pvvx/ATC_MiThermometer #5: Strange readings

rjblake, 2021-01-02:
I used your .bin file to update and now I get very strange readings compared to Aaron's. xx:xx:xx is your firmware and yy:yy:yy is Aaron's. Both are configured to used Custom and not MiLike and details on screen are showing correctly. Any thoughts? 

2021-01-02 14:13:44 - Device: A4:C1:38:xx:xx:xx Temp: 1280.8c Humidity: 253% Batt: 16%
2021-01-02 14:13:49 - Device: A4:C1:38:yy:yy:yy Temp: 20.6c Humidity: 48% Batt: 92%

pvvx, 2021-01-02:
Firmware Ver 0.8, Advertising data:
```
struct _packed_ {
	uint8_t		size;	// = 18
	uint8_t		uid;	// = 16, 16-bit UUID
	uint16_t	UUID;	// = 0xfe95, UUID: 'fe95'
	int16_t		temperature; // x 0.01 degree
	int16_t		humidity; // x 0.01 %
	uint16_t	battery_mv; // mV
	uint8_t		battery_level; // %
	uint8_t		counter; // measurement number
	uint8_t		flags; 
} adv_custom_t;
```

rjblake, 2021-01-02:
Thanks - didn't realise that you'd changed the structure of the advertising. Will need to modify my code accordingly

ralf-e, 2021-01-02:
does that mean the FW is no longer compatible with Tasmota BLE or similar?
I ask before I update ...

pvvx, 2021-01-02:
Advertising Mi
```
// 16-bit UUID for Members 0xFE95 Xiaomi Inc. https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf
typedef struct __attribute__((packed)) _adv_mi_t {
	uint8_t		size;	// = 21
	uint8_t		uid;	// = 0x16, 16-bit UUID https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/
	uint16_t	UUID;	// = 0xFE95, 16-bit UUID for Members 0xFE95 Xiaomi Inc.
	uint16_t	flg;	//
	uint16_t    dev_id; //
	uint8_t		counter; // measurement number
	uint8_t		MAC[6];
	// +15: 0x0A, 0x10, 0x01, t_lv, 0x02, b_lo, b_hi
	// +15: 0x0D, 0x10, 0x04, t_lo, t_hi, h_lo, h_hi
	uint8_t     data_id; 	// = 0x0A or 0x0D
	uint8_t     nx10; 		// ? = 0x10
	union {
		struct {
			uint8_t		id1; // = 0x01
			uint8_t		battery_level; // 0..100 %
			uint8_t		id2; // = 0x02
			uint8_t		battery_mv;
		}t0a;
		struct {
			uint8_t		id; // = 0x04
			int16_t		temperature; // x 0.01 degree
			uint16_t	humidity; // x 0.01 %
		}t0d;
	};
} adv_mi_t, * padv_mi_t;
```

pvvx, 2021-01-02:
> does that mean the FW is no longer compatible with Tasmota BLE or similar?

Advertising format 'Tasmota BLE' ?

ralf-e, 2021-01-02:
"tasmota BLE MI ESP32" currently receives either ATC-like or MI-like (original FW, encrypted with KEY or unencrypted for LYWSD02), MI-Like with ATC-FW does not work, no data (4. Sensor in the screenshot).
I therefore assume that your current version no longer works with Tasmota either

[image]

pvvx, 2021-01-02:
The firmware revision is not over yet. The Advertising format will change in new versions ...
Current versions have a status - alpha (test only).

pvvx, 2021-01-03:
> "tasmota BLE MI ESP32" currently receives ...

Tasmota is **Proprietary Software**. No spec and open-source. This is not supported. Likewise, there will be no normal support ESP32 - is just another proprietary product.

closed 2021-01-03 by pvvx