pvvx/ATC_MiThermometer #8: Emulate LCD in the WebFlasher

atc1441, 2021-01-06:
Hey, again you made such nice improvements i love it.

I just had the idea that we maybe could ad an canvas with the Thermometer in Web so we can see live what is being displayed on the LCD without the need to go to each one.

its more like a gimmick but i think it would add a lot to the user experience.

it would kind of look like this image https://github.com/atc1441/ATC_MiThermometer/blob/master/Mi_LCD_Segments.jpeg 
but drawn in a canvas based on the status of the LCD from the device, could be even reproduced based on the Notification with the Sensor data

pvvx, 2021-01-07:
**Command 0x60** + [LCD buffer data]:  if LCD buffer data[6] exist -> LCD_flg set bit 2 – LCD shows external data.

- Response and Notify: 0x60,[LCD buffer data]. 

**Command 0x61** + [LCD_flg]:  Setting LCD flags.

- Response: 0x60,  LCD_flg.

**LCD_flg:** bit 1 – Notify on/off,  bit 2 – LCD shows external data

cricri-pingouin, 2021-01-12:
That would be great, then I could save the page on my mobile offline and not need an app at all
*cough* ads *cough*

closed 2021-02-02 by pvvx