pvvx/ATC_MiThermometer #12: OTA flashing takes very long

mley, 2021-01-10:
Thanks for your effort on improving ATC's awesome work even further.

I just updated one of my thermometers to your version 1.1. The previous installed firmware was ATC's custom firmware.

It took around 50 MINUTES to flash the complete firmware. I'm not sure if that's related to the running custom firmware, the web-flasher or something else.

I used a 2017 MBP, Big Sur, Chrome 87 (same setup as when I flashed ATCs firmware). The thermometer had a distance of 30cm to the computer.

I haven't tried flashing the others yet, because I still have to test the new firmware a bit.

Here is the log:
```
Log:
08:56:34: Searching for devices
08:56:47: Connecting to: ATC_7B4211
08:56:50: Detected custom Firmware
08:57:18: File was selected, size: 48816 bytes
08:57:18: Count: 3051
08:57:25: Start DFU
09:46:06: Update done after 2919.46 seconds
09:46:13: Disconnected.
```

Console (last lines only):
```
TelinkMiFlasher.html:609 Status: Sending block nr: 3044 from 3051, 99% done, time since start 2911.411s
TelinkMiFlasher.html:609 OTA: e40bc2a5000004a6000028a600004aa600001a91
TelinkMiFlasher.html:609 Status: Sending block nr: 3045 from 3051, 99% done, time since start 2911.413s
TelinkMiFlasher.html:609 OTA: e50b6aa6000092a6000052a8000082a800008b63
TelinkMiFlasher.html:609 Status: Sending block nr: 3046 from 3051, 99% done, time since start 2911.415s
TelinkMiFlasher.html:609 OTA: e60b52a80000f4a800001ca90000c4a900008b65
TelinkMiFlasher.html:609 Status: Sending block nr: 3047 from 3051, 99% done, time since start 2911.417s
TelinkMiFlasher.html:609 OTA: e70b52a80000cea90000faa900001caa0000c2ff
TelinkMiFlasher.html:609 reading OTA
TelinkMiFlasher.html:609 Status: Sending block nr: 3048 from 3051, 99% done, time since start 2919.451s
TelinkMiFlasher.html:609 OTA: e80b44aa000052a8000068a8000068a8000008af
TelinkMiFlasher.html:609 Status: Sending block nr: 3049 from 3051, 99% done, time since start 2919.454s
TelinkMiFlasher.html:609 OTA: e90b68a80000ffffffffffffffffffffffff30af
TelinkMiFlasher.html:609 Status: Sending block nr: 3050 from 3051, 99% done, time since start 2919.456s
TelinkMiFlasher.html:609 OTA: ea0bffffffff02000000010000002856443ba463
TelinkMiFlasher.html:609 OTA: 02ffea0b15f4
```

ralf-e, 2021-01-10:
this has also happened here sometimes.  the next flash worked mormally again

atc1441, 2021-01-10:
Yes that that happens for me sometimes as well. 
I simply click on reconnect and start the flashing again even while the flash is running

mley, 2021-01-10:
Thank you a lot, @ralf-e @atc1441 . Will try.

closed 2021-01-10 by mley

nadnerb33, 2021-01-30:
I had the same issue with a Macbook Pro and Chrome.
Now I just use my phone and it is always flashes quickly.