pvvx/ATC_MiThermometer #13: xiaomi-thb.ino and temperature below 0

fulvioalessio, 2021-01-13:
Hi, I've tested 1.2 Fw on my devices.
Seems that esp32 code doesn't manage negative temperature.

Here .ino debug:

`[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x16 (), length: 17, data: 1a181477c638c1a4fcffda1cd10a3f9d0d

[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 10, data: 4154435f433637373134

[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: ATC_C67714

Found service '181a' data len: 19, 12161a181477c638c1a4fcffda1cd10a3f9d0d

MAC: 5 Temp: **655.32°**, Humidity: 73.86%, Vbatt: 2769, Battery: 63%, flg: 0x0d, cout: 157
FEED: MAC a4c138c67714
 temp **655.320007** humidity 73.860001 bat 2769.000000

Found service '181a' data len: 19, 12161a181477c638c1a4f4ff041dcb0a3eaf0d

MAC: 5 Temp: **655.24°**, Humidity: 74.28%, Vbatt: 2763, Battery: 62%, flg: 0x0d, cout: 175
FEED: MAC a4c138c67714
 temp **655.239990** humidity 74.279999 bat 2763.000000
`

tomcat-xx, 2021-01-14:
LYWSD03MMC is specified for temperatures above 0 only (0-60). So maybe it's not an issue of reading software. I have this for 3 different ways of accessing LYWSD03MMC data. On the other hand I often read about some python script as base of coding. So if there's an error maybe it is inherited to all other implementations.

fulvioalessio, 2021-01-14:
Hi,

LYWSD03MMC can read temperature below 0. I think there is only a little issue in .ino source code.
With atc1441 firmware I was able to display correct temperature.
With new firmware I used .ino example to understand new data format and you can view the result.

pvvx, 2021-01-14:
float temp = *(int16_t*)(serviceData + 10) / 100.0;

https://github.com/pvvx/ATC_MiThermometer/commit/bb5c031d2816bca1818c2b01a63e1c472368e546

Sensor SHTV3/SHTC3: -40°C to 125°C.
```
20:20:39.725 -> Found service '181a' data len: 19, 12161a18a1cb7f38c1a4dbfdb00e8f0a37d80d
20:20:39.725 -> MAC: a4c1387fcba1
20:20:39.725 -> Temp: -5.49°, Humidity: 37.60%, Vbatt: 2703, Battery: 55%, flg: 0x0d, cout: 216
```

tomcat-xx, 2021-01-14:
Great to hear (both answers above ;-)

pvvx, 2021-01-14:
Advertising in 4 formats:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/ble.h#L31-L88
```
// GATT Service 0x181A Environmental Sensing
// All data little-endian
typedef struct __attribute__((packed)) _adv_custom_t {
	uint8_t		size;	// = 19
	uint8_t		uid;	// = 0x16, 16-bit UUID
	uint16_t	UUID;	// = 0x181A, GATT Service 0x181A Environmental Sensing
	uint8_t		MAC[6]; // [0] - lo, .. [6] - hi digits
	int16_t		temperature; // x 0.1 degree
	uint16_t	humidity; // x 0.01 %
	uint16_t	battery_mv; // mV
	uint8_t		battery_level; // 0..100 %
	uint8_t		counter; // measurement count
	uint8_t		flags; 
} adv_custom_t, * padv_custom_t;

// GATT Service 0x181A Environmental Sensing
// mixture of little-endian and big-endian!
typedef struct __attribute__((packed)) _adv_atc1441_t {
	uint8_t		size;	// = 16
	uint8_t		uid;	// = 0x16, 16-bit UUID
	uint16_t	UUID;	// = 0x181A, GATT Service 0x181A Environmental Sensing (little-endian)
	uint8_t		MAC[6]; // [0] - hi, .. [6] - lo digits (big-endian!)
	uint8_t		temperature[2]; // x 0.1 degree (big-endian!)
	uint8_t		humidity; // x 1 %
	uint8_t		battery_level; // 0..100 %
	uint8_t		battery_mv[2]; // mV (big-endian!)
	uint8_t		counter; // measurement count
} adv_atc1441_t, * padv_atc1441_t;

// UUID for Members 0xFE95 Xiaomi Inc. https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf
// All data little-endian, + https://github.com/Magalex2x14/LYWSD03MMC-info
typedef struct __attribute__((packed)) _adv_mi_t {
	uint8_t		size;	// = 21
	uint8_t		uid;	// = 0x16, 16-bit UUID https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/
	uint16_t	UUID;	// = 0xFE95, 16-bit UUID for Members 0xFE95 Xiaomi Inc.
	uint16_t	ctrl;	// = 0x3050 Frame ctrl
	uint16_t    dev_id; // = 0x055B	Device type
	uint8_t		counter; // 0..0xff..0 measurement count
	uint8_t		MAC[6];	// [0] - lo, .. [6] - hi digits
	// +15: 0x0A, 0x10, 0x01, t_lv, 0x02, b_lo, b_hi
	// +15: 0x0D, 0x10, 0x04, t_lo, t_hi, h_lo, h_hi
	uint8_t     data_id; 	// = 0x0A or 0x0D
	uint8_t     nx10; 		// = 0x10
	union {
		struct {
			uint8_t		len1; // = 0x01
			uint8_t		battery_level; // 0..100 %
			uint8_t		len2; // = 0x02
			uint16_t	battery_mv;
		}t0a;
		struct {
			uint8_t		len; // = 0x04
			int16_t		temperature; // x0.1 C
			uint16_t	humidity; // x0.1 %
		}t0d;
	};
} adv_mi_t, * padv_mi_t;
```

pvvx, 2021-01-14:
[image]

closed 2021-02-07 by pvvx