pvvx/ATC_MiThermometer #14: C401 fw - error correction

michapr, 2021-01-16:
Hi,

I'm testing the latest version (12) for 401 - it is working great, thanks!
Energy usage is minimal, so now the device can be used as expected ;)

Only one thing - I have got within the last 24hours 3 wrong results of temp/hum (>200 each).
Had it never before with atc or your older fw.
Possible any wakeup issue?

Maybe you can add a simple check before sending? temp <100 and hum<100 ?
I think it is a simple way...

Thanks again!

Michael

PS: I would recommend to set the comfort smiley as default in the bin ;)

pvvx, 2021-01-17:
> Only one thing - I have got within the last 24hours 3 wrong results of temp/hum (>200 each).

? Temperature sensor CRC check enabled.

michapr, 2021-01-17:
Yes,, I'm not about CRC error.
I think the sensor itself give back a wrong value after wakeup.

closed 2021-02-02 by pvvx