pvvx/ATC_MiThermometer #17: Domoticz

Paroxxx, 2021-01-17:
Hello,

What a great job, thanks !!
Not an issue but a request.

I try to use [https://github.com/ClementMarrast/mijia-sensor-domoticz]. It works with original firmware but not with yours. It can't connect to the device.

I think there is something to change in the code in "mitemp2_bt_poller.py" file like "_HANDLE_READ_BATTERY_LEVEL = 0x001B" and in "domoticz_mijaa.py" but i'm completely newbie...

Could you help to find the correct changes ?

rjblake, 2021-01-29:
Take a look at this instead [https://github.com/JsBergbau/MiTemperature2](https://github.com/JsBergbau/MiTemperature2). You can also use for multiple Thermometers in Domoticz through callback and it listens for advertising rather than polling, so better battery life overall.

Paroxxx, 2021-01-31:
I Will have a look, thanks !

closed 2021-02-02 by pvvx