pvvx/ATC_MiThermometer #19: nodered function

podarok, 2021-01-21:
```javascript
var payload = msg.payload;

if(payload.address.startsWith("a4:c1:38") &&
  payload.advertisement.localName.startsWith("ATC_")) {
    var buffer = payload.advertisement.serviceData[0].data;
    var mac = payload.address

  
    if(buffer.length >= 15 && payload.advertisement.serviceData[0].uuid == "181a") { // pvvx
        msg.payload = {
        "mac": mac,
        "temperature": buffer.readIntLE(6, 2) / 100.0,
        "humidity": buffer.readIntLE(8, 2) / 100.0,
        "battery": buffer.readIntLE(12, 1),
        "vbat": buffer.readIntLE(10, 2)/ 1000.0
    };	
       return msg;	       
    }
}
return null;
```

pvvx, 2021-01-21:
https://pvvx.github.io/ATC_MiThermometer/Advertising.html
```
22:42:26: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05271C250F38C1A40D10046D00E301
22:42:26: MAC: A4C1380F251C, Count: 39, Temp: 10.9°C, Humi: 48.3%
22:42:36: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05281C250F38C1A40D10046D00E101
22:42:36: MAC: A4C1380F251C, Count: 40, Temp: 10.9°C, Humi: 48.1%
22:42:46: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05291C250F38C1A40D10046C00E001
22:42:46: MAC: A4C1380F251C, Count: 41, Temp: 10.8°C, Humi: 48%
22:42:56: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B052A1C250F38C1A40D10046C00DF01
22:42:56: MAC: A4C1380F251C, Count: 42, Temp: 10.8°C, Humi: 47.9%
22:43:03: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B052B1C250F38C1A40A10014502060B
22:43:03: MAC: A4C1380F251C, Count: 43, Bat: 69%, Vbat: 2822 mV
22:43:23: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B052D1C250F38C1A40A100145020B0B
22:43:23: MAC: A4C1380F251C, Count: 45, Bat: 69%, Vbat: 2827 mV
22:43:33: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B052E1C250F38C1A40A100145020D0B
22:43:33: MAC: A4C1380F251C, Count: 46, Bat: 69%, Vbat: 2829 mV
22:43:41: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, 1C250F38C1A43B04EA120B0B452F0D
22:43:41: MAC: A4C1380F251C, Bat: 69%, Vbat: 2827 mV , Temp: 10.83°C, Humi: 48.42%, Count: 47, Flg: 13
22:43:48: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, A4C1380F251C006C31450B0930
22:43:48: MAC: A4C1380F251C, Bat: 69%, Vbat: 2825 mV, Temp: 10.8°C, Humi: 49%, Count: 48
22:44:01: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, 1C250F38C1A44004C3130B0B45310D
22:44:01: MAC: A4C1380F251C, Bat: 69%, Vbat: 2827 mV , Temp: 10.88°C, Humi: 50.59%, Count: 49, Flg: 13
22:44:11: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, 1C250F38C1A4470409140B0B45320D
22:44:11: MAC: A4C1380F251C, Bat: 69%, Vbat: 2827 mV , Temp: 10.95°C, Humi: 51.29%, Count: 50, Flg: 13
22:44:11: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, 1C250F38C1A4470409140B0B45320D
22:44:11: MAC: A4C1380F251C, Bat: 69%, Vbat: 2827 mV , Temp: 10.95°C, Humi: 51.29%, Count: 50, Flg: 13
22:44:19: ServiceData Data: 0000181a-0000-1000-8000-00805f9b34fb, A4C1380F251C006E33450B0C33
22:44:19: MAC: A4C1380F251C, Bat: 69%, Vbat: 2828 mV, Temp: 11°C, Humi: 51%, Count: 51
22:44:36: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05341C250F38C1A40D10046E000902
22:44:36: MAC: A4C1380F251C, Count: 52, Temp: 11°C, Humi: 52.1%
22:44:54: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05361C250F38C1A40A10014502070B
22:44:54: MAC: A4C1380F251C, Count: 54, Bat: 69%, Vbat: 2823 mV
22:45:04: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B05371C250F38C1A40A100146020E0B
22:45:04: MAC: A4C1380F251C, Count: 55, Bat: 70%, Vbat: 2830 mV
```

pvvx, 2021-01-21:
> if(payload.address.startsWith("a4:c1:38")
>  payload.advertisement.localName.startsWith("ATC_")

Get/Set dev. name:
[image]

Get/Set dev. MAC[s]:
[image]

closed 2021-02-07 by pvvx