pvvx/ATC_MiThermometer #21: Battery level on MI advertising mode

marcogpg, 2021-01-23:
Hi,

First I want to thank your amazing work on this.

I have some of this devices (LYWSD03MMC) on my home connected to HomeAssistant using ESPHome.

I installed this firmware and configured MI advertising mode. 
I get values for temperature and humidity but do not receive updates for battery level.

Looking at https://pvvx.github.io/ATC_MiThermometer/Advertising.html I get this values:

15:32:40: Watching advertisements from "ATC_5FF9B6"...
15:32:58: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B051FB6F95F38C1A40D1004B800B102
15:32:58: MAC: A4C1385FF9B6, Count: 31, Temp: 18.4°C, Humi: 68.9%
15:32:58: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B051FB6F95F38C1A40D1004B800B102
15:32:58: MAC: A4C1385FF9B6, Count: 31, Temp: 18.4°C, Humi: 68.9%
15:33:59: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0525B6F95F38C1A40D1004B700B102
15:33:59: MAC: A4C1385FF9B6, Count: 37, Temp: 18.3°C, Humi: 68.9%
15:33:59: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0525B6F95F38C1A40D1004B700B102
15:33:59: MAC: A4C1385FF9B6, Count: 37, Temp: 18.3°C, Humi: 68.9%
15:35:49: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0530B6F95F38C1A40D1004B800B102
15:35:49: MAC: A4C1385FF9B6, Count: 48, Temp: 18.4°C, Humi: 68.9%
15:35:49: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0530B6F95F38C1A40D1004B800B102
15:35:49: MAC: A4C1385FF9B6, Count: 48, Temp: 18.4°C, Humi: 68.9%
15:36:10: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0532B6F95F38C1A40D1004B800B102
15:36:10: MAC: A4C1385FF9B6, Count: 50, Temp: 18.4°C, Humi: 68.9%
15:36:10: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0532B6F95F38C1A40D1004B800B102
15:36:10: MAC: A4C1385FF9B6, Count: 50, Temp: 18.4°C, Humi: 68.9%
15:36:27: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0534B6F95F38C1A40D1004B800B102
15:36:27: MAC: A4C1385FF9B6, Count: 52, Temp: 18.4°C, Humi: 68.9%
15:36:27: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0534B6F95F38C1A40D1004B800B102
15:36:27: MAC: A4C1385FF9B6, Count: 52, Temp: 18.4°C, Humi: 68.9%
15:36:57: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0537B6F95F38C1A40D1004B800B102
15:36:57: MAC: A4C1385FF9B6, Count: 55, Temp: 18.4°C, Humi: 68.9%
15:36:57: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0537B6F95F38C1A40D1004B800B102
15:36:57: MAC: A4C1385FF9B6, Count: 55, Temp: 18.4°C, Humi: 68.9%
15:37:17: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0539B6F95F38C1A40D1004B800B102
15:37:17: MAC: A4C1385FF9B6, Count: 57, Temp: 18.4°C, Humi: 68.9%
15:37:17: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B0539B6F95F38C1A40D1004B800B102
15:37:17: MAC: A4C1385FF9B6, Count: 57, Temp: 18.4°C, Humi: 68.9%
15:37:40: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B053BB6F95F38C1A40D1004B800B102
15:37:40: MAC: A4C1385FF9B6, Count: 59, Temp: 18.4°C, Humi: 68.9%
15:37:40: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B053BB6F95F38C1A40D1004B800B102
15:37:40: MAC: A4C1385FF9B6, Count: 59, Temp: 18.4°C, Humi: 68.9%
15:37:43: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B053BB6F95F38C1A40D1004B800B102
15:37:43: MAC: A4C1385FF9B6, Count: 59, Temp: 18.4°C, Humi: 68.9%
15:37:43: ServiceData Data: 0000fe95-0000-1000-8000-00805f9b34fb, 50305B053BB6F95F38C1A40D1004B800B102
15:37:43: MAC: A4C1385FF9B6, Count: 59, Temp: 18.4°C, Humi: 68.9%

Shouldn't be there advertising data for battery also?

Thanks again!

Marco

pvvx, 2021-01-23:
V1.6: Fix alternation of Advertising in mi mode.

marcogpg, 2021-01-24:
It's working now. 

Thanks!

closed 2021-01-24 by marcogpg