pvvx/ATC_MiThermometer #23: No issue, need explanation regarding intervals and battery consumption

nortuzar, 2021-01-24:
Hi, I flashed the software on 3 LYWSD03MMC. Its working well, but Im not an expert so Im kind of lost when trying to optimize life battery.

- Advertising interval is 2500, this is milliseconds?  Can I increase this as much as I want? Like I only need 5 minutes updates

- What happen when the option Sensor in "Low Power mode" is active?

- I understand The RF TX Power is the strength of the signal, can I lower it making sure the signal is received by my ble host? (rasp w)

Thanks for the help, nice work!

pvvx, 2021-01-25:
> * Advertising interval is 2500, this is milliseconds?  Can I increase this as much as I want? Like I only need 5 minutes updates

Max 10 sec (10000 ms). Not all devices can connect if parameter above 3 seconds.

> * What happen when the option Sensor in "Low Power mode" is active?

Less consumption, less accurate temperature and humidity readings.

> * I understand The RF TX Power is the strength of the signal, can I lower it making sure the signal is received by my ble host? (rasp w)

Very little dependence on energy consumption.

pvvx, 2021-01-25:
RF TX Power: 0 dB
Advertising interval: 10000 (10 sec)
Measure interval: 10 (100 sec)
LCD not show battery.

The expense is reduced by 53.2% of the default.

nortuzar, 2021-01-25:
> 
> 
> RF TX Power: 0 dB
> Advertising interval: 10000 (10 sec)
> Measure interval: 10 (100 sec)
> LCD not show battery.
> 
> The expense is reduced by 53.2% of the default.

Great, using the last fw 1.6, set it up as you said and disable the "Low power mode" to gather better accurate readings, Like this should be ok?
[image]

Thanks for the great firmware

pvvx, 2021-01-25:
What are you saving?
By default, if CR2032 is 200mAh, only change the beacon interval:
[image]

MacSass, 2021-02-11:
Hi @pvvx ,
can you quantify power usage for showing battery, vs. not showing battery on LCD? How much would it "cost" showing battery level vs. default settings?

Also, when I increase the "Minimum LCD refresh rate", to e.g. 5 seconds - does it mean LCD only refreshes every 5 seconds?

Would that have a beneficial effect for battery life? Does refresh need much more power compared to just "show" something on LCD? As LYWSD03MMC does not have e-paper but standard LCD I would think that "showing" would be similar to "refresh" (while real e-paper only needs power when refreshing)?

Thanks for your help getting the most out of those small devices ...

pvvx, 2021-02-12:
> can you quantify power usage for showing battery, vs. not showing battery on LCD? How much would it "cost" showing battery level vs. default settings?

Battery level measurement no more than 200 μs at 4 mA
The default battery measurement cycle is 10 seconds (10,000,000 μs).
Average consumption: (200 * 4) / 10,000,000 = 0.00008 mA = max(!) 80 nA 

> Also, when I increase the "Minimum LCD refresh rate", to e.g. 5 seconds - does it mean LCD only refreshes every 5 seconds?

[image]
The LCD is updated every 5 seconds (5000ms),
LCD refresh procedure takes less than 1ms at 4mA:
Average consumption:  (1 * 4) / 5000 = 0.0008 mA = max(!) 0.8 μA

The rest of the equipment at default settings consumes less than 15 μA
CR2032 = 200+ mAh.
 
> Would that have a beneficial effect for battery life? Does refresh need much more power compared to just "show" something on LCD? As LYWSD03MMC does not have e-paper but standard LCD I would think that "showing" would be similar to "refresh" (while real e-paper only needs power when refreshing)?

LCD consumption is less than EPD (Electronic-ink).
E-ink requires increased voltage for renewal and the consumption for DC-DC is quite large.
https://pvvx.github.io/MHO_C401/power_altfw.html

MacSass, 2021-02-12:
Thanks a lot - looks like I don´t have to worry about LCD update cycle then and not about battery measurement ... so I guess I will leave all that at default values.
Eventually I will look at replacing the CR2032 with 2xAA Batteries where the sensor is hidden anyways - that should give really long life ...

Thanks a lot for our deep analysis and inputs.

pvvx, 2021-02-14:
https://pvvx.github.io/ATC_MiThermometer/CustPower.html

jds11111, 2021-03-01:
Am trying to decide if it is worth reflashing my 14 devices with your new firmware, instead of Aaron's. Did you happen to compare power consumption with that firmware?

penright, 2021-03-03:
If the Advertising interval is set to 2500, 2.5 seconds, then should I see an advertisement in ESPHome every 2.5 seconds?

I am including some of the detail, I cut out all the polling that did not have anything to do with the ATC.
The summary of when the ATC advertised are 16:04:40, 16:05:06, 16:07:36, 16:09:07, 16:12:28, and etc
There are from 30 seconds to 3-minute gaps. Is there something more than the "Advertising interval" that controls when it actually sends something?

**_Deatil_**
```
[16:04:38][D][esp32_ble_tracker:544]: Found device 08:28:61:A9:8A:47 RSSI=-91
[16:04:38][D][esp32_ble_tracker:565]:   Address Type: RANDOM
...
[16:04:40][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:04:40][D][esp32_ble_tracker:569]:   TX Power: 2
[16:04:40][D][esp32_ble_tracker:544]: Found device A4:C1:38:4F:19:3D RSSI=-50
[16:04:40][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[16:04:40][D][esp32_ble_tracker:567]:   Name: 'ATC_4F193D'
[16:04:41][D][esp32_ble_tracker:544]: Found device 6A:4D:82:BA:7D:48 RSSI=-93
[16:04:41][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:04:41][D][esp32_ble_tracker:569]:   TX Power: 2
...
[16:05:06][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:05:11][D][esp32_ble_tracker:544]: Found device 0B:38:93:67:D5:C3 RSSI=-86
[16:05:11][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:05:36][D][atc_mithermometer:116]: Got ATC MiThermometer (A4:C1:38:4F:19:3D):
[16:05:36][D][atc_mithermometer:119]:   Temperature: 24.2 °C
[16:05:36][D][atc_mithermometer:122]:   Humidity: 20 %
[16:05:36][D][atc_mithermometer:125]:   Battery Level: 86 %
[16:05:36][D][atc_mithermometer:128]:   Battery Voltage: 2.977 V
[16:05:36][D][sensor:092]: 'ATC Temperature (Shop Wall)': Sending state 24.20000 °C with 1 decimals of accuracy
[16:05:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:05:56'
[16:05:36][D][sensor:092]: 'ATC Humidity (Shop Wall)': Sending state 20.00000 % with 0 decimals of accuracy
[16:05:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:05:56'
[16:05:36][D][sensor:092]: 'ATC Battery-Level (Shop Wall)': Sending state 86.00000 % with 0 decimals of accuracy
[16:05:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:05:56'
[16:05:36][D][sensor:092]: 'ATC Battery-Voltage (Shop Wall)': Sending state 2.97700 V with 3 decimals of accuracy
[16:05:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:05:56'
[16:06:20][D][esp32_ble_tracker:544]: Found device 41:C3:A2:AF:07:FF RSSI=-92
[16:06:20][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:06:20][D][esp32_ble_tracker:569]:   TX Power: 2
[16:06:27][D][esp32_ble_tracker:544]: Found device 5C:74:DE:78:39:37 RSSI=-90
[16:06:27][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:06:34][D][esp32_ble_tracker:544]: Found device 3B:8F:80:FD:DA:B0 RSSI=-93
[16:06:34][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:07:36][D][atc_mithermometer:116]: Got ATC MiThermometer (A4:C1:38:4F:19:3D):
[16:07:36][D][atc_mithermometer:119]:   Temperature: 24.2 °C
[16:07:36][D][atc_mithermometer:122]:   Humidity: 20 %
[16:07:36][D][atc_mithermometer:125]:   Battery Level: 86 %
[16:07:36][D][atc_mithermometer:128]:   Battery Voltage: 2.976 V
[16:07:36][D][sensor:092]: 'ATC Temperature (Shop Wall)': Sending state 24.20000 °C with 1 decimals of accuracy
[16:07:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:07:56'
[16:07:36][D][sensor:092]: 'ATC Humidity (Shop Wall)': Sending state 20.00000 % with 0 decimals of accuracy
[16:07:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:07:56'
[16:07:36][D][sensor:092]: 'ATC Battery-Level (Shop Wall)': Sending state 86.00000 % with 0 decimals of accuracy
[16:07:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:07:56'
[16:07:36][D][sensor:092]: 'ATC Battery-Voltage (Shop Wall)': Sending state 2.97600 V with 3 decimals of accuracy
[16:07:36][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:07:56'
[16:08:59][D][esp32_ble_tracker:544]: Found device 4C:30:F7:B8:1B:2C RSSI=-88
[16:08:59][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:08:59][D][esp32_ble_tracker:569]:   TX Power: 2
[16:09:07][D][atc_mithermometer:116]: Got ATC MiThermometer (A4:C1:38:4F:19:3D):
[16:09:07][D][atc_mithermometer:119]:   Temperature: 24.2 °C
[16:09:07][D][atc_mithermometer:122]:   Humidity: 20 %
[16:09:07][D][atc_mithermometer:125]:   Battery Level: 86 %
[16:09:07][D][atc_mithermometer:128]:   Battery Voltage: 2.975 V
[16:09:07][D][sensor:092]: 'ATC Temperature (Shop Wall)': Sending state 24.20000 °C with 1 decimals of accuracy
[16:09:07][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:09:27'
[16:09:07][D][sensor:092]: 'ATC Humidity (Shop Wall)': Sending state 20.00000 % with 0 decimals of accuracy
[16:09:07][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:09:27'
[16:09:07][D][sensor:092]: 'ATC Battery-Level (Shop Wall)': Sending state 86.00000 % with 0 decimals of accuracy
[16:09:07][D][text_sensor:015]: 'ATC Heartbeat (Shop Wall)': Sending state '03/03/2021 22:09:27'
[... 123 lines cut ...]
[16:19:11][D][atc_mithermometer:128]:   Battery Voltage: 2.977 V
[16:19:11][D][sensor:092]: 'ATC Temperature (Shop Wall)': Sending state 24.20000 °C with 1 decimals of accuracy
[... cut ...]
[16:19:35][D][esp32_ble_tracker:148]: Starting scan...
[16:19:35][D][esp32_ble_tracker:544]: Found device 0A:91:B2:A2:30:9C RSSI=-79
[16:19:35][D][esp32_ble_tracker:565]:   Address Type: RANDOM
...
[16:20:45][D][esp32_ble_tracker:544]: Found device 6D:1F:2F:32:2C:68 RSSI=-92
[16:20:45][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[16:20:45][D][esp32_ble_tracker:569]:   TX Power: 2
[16:21:02][D][esp32_ble_tracker:544]: Found device 57:8F:2E:73:E8:80 RSSI=-92
[16:21:02][D][esp32_ble_tracker:565]:   Address Type: RANDOM

```

MichielioZ, 2021-03-04:
@penright Please just attach a file or provide a link to a file with this output. Right now it's so long, I think people will not read anything underneath it. Besides: the data only confirms that you were not lying about what you said, then again, nobody assumed you were lying...

@pvvx I use your custom firmware, so thank you for that. While I admire your expertise on this subject and can appreciate you making all the difficult calculations, as a user I just want simple settings...
I think most people will want the longest battery life possible, but have no idea how to use the settings, as they are too technical.
You mentioned earlier:
`RF TX Power: 0 dB
Advertising interval: 10000 (10 sec)
Measure interval: 10 (100 sec)
LCD not show battery.

The expense is reduced by 53.2% of the default.`

Did I read this right ? Do these settings make the battery last a LOT longer compared to the default settings ?
If so, why are these not the default ?

penright, 2021-03-04:
As per @MichielioZ suggestion, here is the output of ESPHome log showing the Bluetooth advertisements received.
I was expecting to see a regular advertisement for the device (ATC_4F193D). Does it only advertise when a value has changed?

Trying to detect a failed or not responding device is what got me started down this path. Since from ESPHome perspective, the event is random. So I thought I would use HA last_updated state attribute. Looks like HA only updates the state if the state changed. So the question becomes what is the max time to allow for a temp change before declaring the ATC device missing. So I created a sensor that would be updated on every received advertisement. There are two ways, one is using the "on_value" event or baking it into the "atc_mithermometer" component. That is when I noticed the irregularity of the advertisement. I was searching for information and ran across this ticket. I thought the question would fit, but by adding the "and battery consumption" maybe I should open a new ticket?

The only explanation I can think of, ESPHome missing an advertisement, or ATC has some algorithm that skips trying to save battery. Understanding that will help set the "max" receive timeout.

[ESPHome_Log_.txt](https://github.com/pvvx/ATC_MiThermometer/files/6084494/ESPHome_Log_.txt)

MichielioZ, 2021-03-04:
@penright I use the "Passive BLE monitor" (ble_monitor) integrations in Home Assistant and with the default settings of the firmware I get a regular update interval of 1 minute. Sometimes it skips one, but that happens rarely. Maybe it's an ESPHome problem you are having ?

pvvx, 2021-03-14:
> Did I read this right ? Do these settings make the battery last a LOT longer compared to the default settings ?
> If so, why are these not the default ?

The increased advertising interval makes it impossible to connect to the device with old BT hardware, Apple and other poorly written drivers and systems.
The default (2.5 seconds) is already on the verge of being compatible with older BT hardware.
Apple's recommendations say up to 2 seconds, and according to BLE standards, up to 10 seconds.
The default (2.5 sec) is already on the verge of compatibility with older BT hardware.

MichielioZ, 2021-03-14:
@pvvx Ah, ok, that explains why it isn't the default, but 53.2% better power usage is still worth mentioning somewhere...
My phone has BT 5.0, so pretty new version.
Is it safe to try out loading these "optimal" settings with my phone ?

In daily use I use them with my Raspberry Pi 3B onboard BLE (BCM43438 chip).
If they don't work with my Pi I should be able to safely return the old firmware with my phone, right ?

pvvx, 2021-03-16:
> If they don't work with my Pi I should be able to safely return the old firmware with my phone, right ?

If it will be possible to connect to the device for which you set the interval of 10 seconds :)

MichielioZ, 2021-03-16:
> 
> 
> > If they don't work with my Pi I should be able to safely return the old firmware with my phone, right ?
> 
> If it will be possible to connect to the device for which you set the interval of 10 seconds :)
> 
Yes, that's why I asked if my Phone (with Bluetooth 5.0, so newer version) could still connect if I set the interval to 10 seconds.
That way I could test it with my Raspberry Pi and if it didn't work, go back to the old settings...
Guess I will just try my luck with 1, no pain no gain ;-)

I understand it is hard for you to try and satisfy everyone.
Also to cover all implementations of Bluetooth seems impossible.
But if the difference in settings can be 1 month battery and up to 1 year battery, we just have to test and let people know what works...

If for example I find that I can safely test up to 10 secs interval with my Phone
(Qualcomm SDM730 Snapdragon 730, Bluetooth 5.0, A2DP, LE) we can let other people know it is safe to test with similar hardware...
ESP32 boards may be quirky when it comes to Bluetooth, but it doesn't matter if I have a Phone or another device that I can use to adjust the firmware settings...
I will let you (and others) know my experiences after I know the limitations of my Phone (I have some ESP32 boards to test with).

MichielioZ, 2021-03-16:
Testing update:

For the time being I only tested with the Advertising interval and Measure interval combination.
I didn't touch any other settings except I hide the annoying smiley ;-)
So I did not set RX TX Power to 0, but left it at VANT +3.01 dbm.
A/M of 5000/5: working, no further testing done yet.
A/M of 8000/8: working, no further testing done yet.
A/M of 10000/10: slow to connect and update, but working.
Last setting also works with my Raspberry Pi 3B but it doesn't pick up most of the broadcasts at 3 meters distance.
In Home Assistant the update interval seems to be between 10 and 20 minutes, so yeah, not great...
Going to go back 1 step and test some more... I'll be back ;-)

pvvx, 2021-03-17:
> But if the difference in settings can be 1 month battery and up to 1 year battery, we just have to test and let people know what works...

The default settings should be comparable to the original firmware in terms of consumption.
At current default settings, consumption is less than that of the original.

pvvx, 2021-03-17:
Testing for the number of drops in receiving Advertising packages from the sensor.
[The default settings are 10 meters to the sensor, the radio environment is noisy - more than 3 thousand third-party beacons per second, several WiFi routers.](https://github.com/pvvx/ATC_MiThermometer/issues/66#issuecomment-800056658)

Any USB-BT adapter - not received packets > 90%.
Any Sniffer - about 5% not received packets due to collisions on the radio air from nearby sources (mainly WiFi routers).

pvvx, 2021-03-17:
@MichielioZ  
> I understand it is hard for you to try and satisfy everyone.
> Also to cover all implementations of Bluetooth seems impossible.

There are sources-code for any case.
The my license does not prohibit copying, distributing and modify parts of the project at your discretion, even without pointing out the source from me. Reference to other sources should be requested from the authors. Create your repository and describe what you like there... 
I do not have the opportunity to test all the options and support this project for more than a couple of months.

closed 2022-05-01 by pvvx