pvvx/ATC_MiThermometer #24: Blank screen after flashing version 1.6

migreg, 2021-01-24:
Hi,

I don't know if I'm missing something. 

Just tested version 1.6 of your firmware, coming from an ATC firmware.

Screen is blank although I get data in the browser and can set things. 

I reflashed ATC latest FW back again and works fine. Did it twice to double check and I get consistent results. No data shown with your firmware. Also speed when reflasing is very low if protocol is set to all. Setting it to MI flashes quickly.

Thanks

pvvx, 2021-01-25:
The page contains links to two types of firmware for different devices.
**LYWSD03MMC** and **MHO-C401**

If you have LYWSD03MMC, then choose the firmware for it.

migreg, 2021-01-25:
OK my fault then, sorry! Thanks!

closed 2021-01-31 by pvvx