pvvx/ATC_MiThermometer #25: Unable to connect anymore to LYWSD03MMC after changing Advertising interval

csr40, 2021-01-25:
Hello,

I set "Advertising interval" to 10000 and now I'm unable to connect anymore to the thermometer (I did it on 2 devices).
Is there any way to reset settings of the thermometer with firmware version 1.6? I tried to short reset pins but with no effect.

Thank in advance for your help,
Best Regards

pvvx, 2021-01-25:
[image]

Windows 10 + USB-BT very low cost adapter CSR8510 (A10,,, MSFT100)
Drivers:
[image]
```
Log:
13:25:32: Searching for devices
13:26:17: Connecting to: ATC_0B5EED
13:26:25: NetworkError: Connection failed for unknown reason.
13:26:25: Reconnect 1 from 5
13:26:41: Detected custom Firmware
13:26:41: Hardware Version: LYWSD03MMC.
13:26:41: Software Version: 1.6, Custom config: [6, 0, 0, 0, 160, 4, 191, 124, 49]
```
160 * 62.5 = 10000 ms

LowCost Android Phone:
```
Log:
13:38:24: Searching for devices
13:38:39: Connecting to: ATC_0B5EED
13:39:06: Detected custom Firmware
13:39:06: Hardware Version: LYWSD03MMC.
13:39:06: Software Version: 1.6, Custom config: [6, 0, 0, 0, 160, 4, 191, 124, 49]
```

pvvx, 2021-01-25:
nRF Connect.
Set default settings:
[image]
[image]

csr40, 2021-01-25:
Thank you very much Victor!
I used nRF (android phone) with your tips and now I can connect to my thermometers again.

You are great, all the best for you :-)!

closed 2021-01-25 by csr40