pvvx/ATC_MiThermometer #27: proposal: broadcast with historical data

fulvioalessio, 2021-01-25:
Hello pvvx,

I'm working and analyzing data received from 4 thermometers around my house.
I'd like to create a daily graph like GraphAtc, but in different way.

Eg:

`
struct HistoryData {
  int16_t tempMin;
  int16_t tempMax;
  int16_t humMin;
  int16_t humMax;

HistoryData values[24];

`
Data can be shifted every hour:

- values[0] = current hour
- values[1] = previous hour
..

Since I don't have experience with this processor, questions are:

- cpu can manage in memory (no flash) 8*24 bytes ?
- is it possible to manage time like avr ?
- do you think it is possible to create a new broadcast or it is better that server take data when needed ?

Thank you

pvvx, 2021-01-29:
Current memory map:
```
===================================================================
 Section|          Description| Start (hex)|   End (hex)|Used space
-------------------------------------------------------------------
 ramcode|   Resident Code SRAM|           0|        4114|   16660
    text|           Code Flash|        5500|        E868|   37736
  rodata| Read Only Data Flash|        E868|        EF38|    1744
  rtdata|       Retention SRAM|      844114|      8454A4|    5008
      nc|   Wasteful Area SRAM|      8454A4|      845500|      92
   ictag|     Cache Table SRAM|      845500|      845600|     256
  icdata|      Cache Data SRAM|      845600|      845E00|    2048
    data|       Init Data SRAM|      845E00|      845E10|      16
     bss|        BSS Data SRAM|      845E10|      846041|     561
 irq_stk|        BSS Data SRAM|      845E10|      845F90|     384
   stack|       CPU Stack SRAM|      846041|      850000|   40895
   flash|       Bin Size Flash|           0|        EF48|   61256
-------------------------------------------------------------------
Total Free SRAM : 40987 bytes
```
Retention SRAM is 2/3 full. (5008 + 16660) / 32768
You can free about 10 kilobytes without any problem.
As a result, about 20 kilobytes.

pvvx, 2021-01-29:
A specialized version with history logging was supposed. This requires the complication of external software.
A circular buffer of records is placed in Flash.
The unused amount of flash is about 460 kilobytes.
With a record block size of 32 bytes, this gives about 15 thousand records.
When recording every 10 seconds, the total buffer time will be 100 days.

pvvx, 2021-01-29:
> Since I have no experience with this processor

The program is written in С - what does the type of processor have to do with it?

fulvioalessio, 2021-01-29:
> > Since I have no experience with this processor
> 
> The program is written in С - what does the type of processor have to do with it?

Hi,

I'm programming in free time in C with avr & esp32.

> A specialized version with history logging was supposed. This requires the complication of external software.
> A circular buffer of records is placed in Flash.
> The unused amount of flash is about 460 kilobytes.
> With a record block size of 32 bytes, this gives about 15 thousand records.
> When recording every 10 seconds, the total buffer time will be 100 days.
> If you provide OTA and don't make changes to the SDK, cut your free flash in half.
> The record can be compressed and the total number of records will not change...

My idea is to work in ram to avoid flash write and reduce battery life.
About structure, I think that is possible to shrink data in less bytes.
Eg: store humidity min/max in 1 byte: Hmax*100 + Hmin
For temperature, a similar method rounding value min/max to 0.5°C
In case of ram, I think that a broadcast can better. No problem for external software, since the scope maybe a graph on some standalone display (eg: ili9341)

pvvx, 2021-01-29:
Ver 1.9: Recording measurements to flash memory (cyclic buffer for 19632 measurements)

pvvx, 2021-01-29:
[image]

pvvx, 2021-01-30:
> My idea is to work in ram to avoid flash write and reduce battery life.

Memo` used 49 sectors Flash.

If the step of records is 10 seconds, then this is 20,000 records. 139 days. 4.5 months.
Firmware Default Settings: step > 10 sec. (added a random delay for the beacon in BLE).

Let's assume that Flash + CPU write/erase consumption  10 mA (reality less).

Cycle consumption power: 

Sector erase 45 ms.    49*50 = 2.5 sec
Page Program  0.4 ms.    49*0.4 = 19.6 ms
Byte Program 15 us.     49*4096*0.000015 = 3 sec

Total:  3 + 2.5 = **5.5 sec  10 mA**

closed 2021-04-15 by pvvx