pvvx/ATC_MiThermometer #29: Proposal - calculating showing dew point and absolute humidity

cultur98, 2021-01-26:
First of all: What Aaron initiated and you made even more valuable is one of the coolest hack I've seen so far.
Chapeau!
I play a lot with environmental sensors. A very precise measuring SHT85 from Sensirion is in one of my ESP32 gadgets.
Temperature and humidity are shown. I also calculate the dew point and the absolute humidity which is probably of better value than the relative humidity measured by the sensor.
They can easily be calculated from temperature and relative humidity. Wouldn't this be a nice enhancement to show these maybe alternating
[image]
?

pvvx, 2021-01-28:
https://pvvx.github.io/ATC_MiThermometer/DevPoint.html

cultur98, 2021-01-29:
Nice!
You do this with Java Script, right?
My idea was to calculate this in the MiThermometer and show it as an option.
My application is this: I rent an appartement in a house from the 1920s.
The cellar is wet .. you rather find fish than mice in it. .-)
Because of all the bricks in between my cellar, I have no chance for a wireless connection to my appartement.
So a MiThermometer in the cellar showing the dew point and/our absolute humidity would be great!

cultur98, 2021-01-29:
Boooahh .. I just saw that you already .... DID IT in the last update.
Hero!

pvvx, 2021-01-30:
https://pvvx.github.io/ATC_MiThermometer/DevPoint.html
shows the dew point on the LCD device. :)

alexhorner, 2021-02-27:
Please excuse the offtopic nature of this comment - @cultur98 What device are you picturing with M5 embossed on it?

cultur98, 2021-02-28:
Hi Alex, this is a portable thermometer with a SHT85 from Sensirion attached to it.
See:
https://www.sensirion.com/en/environmental-sensors/humidity-sensors/sht85-pin-type-humidity-sensor-enabling-easy-replaceability/

alexhorner, 2021-02-28:
Hey @cultur98 

I was interested in the OLED device itself, rather than the sensor attached, as it looks like it could be quite capable for projects similar to this, having a good display and Micro USB, though thats only at face value

cultur98, 2021-02-28:
Ah, it is an M5 Stick:
https://m5stack.com/products/stick?variant=17054376001626
Seems to be not produced any more.

alexhorner, 2021-02-28:
Ah, thats a shame, thanks!

closed 2022-05-01 by pvvx