pvvx/ATC_MiThermometer #30: [Feature Request] Show time instead of temperature

saikek, 2021-01-27:
Hi,

I'm using LYWSD03MMC mostly as integrations in home-asisstant, so i'm watching for temperature / humidity on dashboards.
It would be much convenient to show time on actual device instead of temperature. 

Is it possible to do ? 

It can be synced with browser manually using TelinkFlasher (or yours) interface.

Thank you.

pvvx, 2021-01-27:
It will be a different device.
There are sensors with watches on sale.

pvvx, 2021-01-29:
An additional time display function has been included since version 1.8.

closed 2021-02-07 by pvvx

vekonyat, 2021-12-22:
Hi, you should use Eclips and the SDK. The most simple for me was to change the code in the lcd.c file. I replaced the code in the show_big_number and show_small_number functions with the code of the show_clock function.
I think you should enable the clock in the web tool as well. Thus it will only show the time.
And you should update the clock with the current time quite often, to show accurate time. If you have an RPI, this can be done with Python and gatttool via BLE. Needs some tinkering though. The nice thing is, if you set zero time with gatttool, you can make a stopwatch, or maybe a timer with some extra coding, now I have no idea how, but it should be possible.
I'm already using these as a display with the method above. (Except the timer)

jpbaril, 2024-06-07:
With all the firmware updates that happened since last post in 2021, is that now something _easily_ possible?
I too would like to use the screen only as a clock.

pvvx, 2024-06-07:
MJWSD05MMC is good for displaying watches.
* Has a built-in clock chip - does not require adjustment. 
* Operates on one battery for over a year.

jpbaril, 2024-06-07:
> MJWSD05MMC is good for displaying watches.
> The [LYWSD02](https://pvvx.github.io/LYWSD02/) 

Thank you for the information.
It's just that I already have the LYWSD03MMC with your firmware. As such I would have liked to repurpose it as a clock and so avoid buying new stuff.