pvvx/ATC_MiThermometer #33: Not working with OpenMQTTGateway

tomcat-xx, 2021-01-29:
Is your version compatible with OpenMQTTGateway? Don't know what exactly is needed. Thought it is just (ATC) advertisement format, but could not get it working. Tried with standard setting and all advertisement formats. "Original" ATC is working just with custom format.

pvvx, 2021-01-30:
OpenMQTTGateway uses HM10 adapter with BT4.0 and AT commands? :)

https://docs.openmqttgateway.com/setitup/ble.html

m-kozlowski, 2021-01-30:
Working for me, on esp32 platform and ATC format.
pvvx's format would require adaptation in main/ZgatewayBT.ino

tomcat-xx, 2021-01-30:
Sorry, did not mention: I'm using ESP32 with integrated BLE module.

And again sorry: Got it running now.

I haad 3 LYWSD03MMC running with atc1441's version for some weeks with OMG. Also when flashing these I could not get it running with yours. Yesterday I got 3 additional ones and tried again with one of it. Without success.

One thing I realized now and maybe overlooked yesterday evening: After flashing device stayed connected. With ATC's firmware/flasher at least it got disconnected when pushing the "Connect" button again. After flashing today I had to switch Bluetooth off at my mobile. Pushing the connect button and even closing browser did not cut the connection. And I think while being connected it does not send advertisements.

Thanks. It's easier when you know it should work ;-)

closed 2021-01-31 by tomcat-xx