pvvx/ATC_MiThermometer #34: Connection dropped with

tikkerei, 2021-01-30:
First of all thank you for you effort and very cool customisation options.

I have following problem, after I have flashed two LYWSD03MMC  via TelinkMiFlasher.html a custom firmawer ATC_Thermometer18.bin. Everything was ok so far, I can also read the data from my home assisntant instance. 
But if I try to connect again to the sensor TelinkMiFlasher.html the connection is dropped after few seconds and I cant do nothing. I have also few sensor with  Custom Firmware Version 1.9, and  the connection is stable.

Is there a workaround to solve this issue? Unfortunately I dont have any hardware for reflashing the sensors.

atc1441, 2021-01-30:
It is possible that this is caused by a nearly empty battery. Can you check with a new one?

tikkerei, 2021-01-30:
Ok finally i reflashed this sensor, i think the dropped connection was because of low battery (35%). After replacing new one, the flashing was done without any issues, and the connection is stable.

tikkerei, 2021-01-30:
> It is possible that this is caused by a nearly empty battery. Can you check with a new one?

exactly, thanks for your reply. I will close this issue.

closed 2021-01-30 by tikkerei

pvvx, 2021-01-30:
1. Сhanged connection intervals?
https://github.com/pvvx/ATC_MiThermometer/issues/25#issuecomment-766739521

2. Entered Pin-code?

tikkerei, 2021-01-30:
> 1. Сhanged connection intervals?
>    [#25 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/25#issuecomment-766739521)
> 2. Entered Pin-code?

Nope, nothing changed. It was with default settings.