pvvx/ATC_MiThermometer #35: What is the max Measure Interval value?

nadnerb33, 2021-01-30:
Ideally I only want a reading every 10 minutes to save battery life of measurement, but also to stop my DB and graphs getting clogged with data.

So I need a measure interval of 240 x 2500ms (default adv interval) = 10 mins

It seems you can only enter 2 digits, so I tried 99 and saved config, but it always resets to 10x
I've also tried 20x and 30x, but it resets. 

Am I doing something wrong? I set the value and then click "Send Config" and the log output says it was successful.

pvvx, 2021-01-30:
> Ideally I only want a reading every 10 minutes to save battery drain from frequent measurements

Increasing the measurement interval will not save energy. Percentage shares.

https://github.com/pvvx/ATC_MiThermometer/issues/23#issuecomment-766898945

pvvx, 2021-01-30:
> but also to stop my DB and graphs getting clogged with data.

Turn on the averaging of results in your database and save every 10 minutes. This will give precision accuracy.

pvvx, 2021-01-30:
For intervals longer than N minute, a different program is required. Incompatible with current firmware.
The device sleeps for N minutes and is activated, measures and issues several beacons. And a new cycle.

nadnerb33, 2021-01-30:
thanks, I'll set it to the max and then filter on the DB side.

closed 2021-01-30 by nadnerb33