pvvx/ATC_MiThermometer #36: Help in configuration

nchieffo, 2021-01-31:
Hello, thanks for this amazing job.
I would like to ask you a question about configuration.
I would like to configure the sensor to:
- send data every 10 mins (I want to limit the bluetooth chip usage if temperature does not change)
- send data if temperature changed 0.2°
- send data if humidity changed 2%
- measure normally to show in the LCD more frequently

Is this something I can do with your firmware?
thanks

pvvx, 2021-01-31:
If the Advertising Interval is increased to more than 2.5 seconds, the connection will not be possible on many devices.
The standard provides for an advertising interval of no more than 10 seconds.
https://github.com/pvvx/ATC_MiThermometer/issues/25#issue-792934062

++ https://github.com/pvvx/ATC_MiThermometer/issues/35#issuecomment-770245674

pvvx, 2021-01-31:
Starting with version 2.0, you can write and read measurement results in Flash. Enable 600 second averaging recording. If necessary, connect and read the last 20,000 stored averages.

nchieffo, 2021-01-31:
Hey Victor, thanks for quick answer.
Does this mean that bluetooth is emitting 'advertising bip' every 2.5 second even if there is no connection (bluetooth icon off)?

I'm asking thiy question because original firmware sends data every 10 mins, so I was thinking the purpose was to save battery, but if bluetooth is constantly emitting, I don't understand why it does not send temperature data too
What's the difference between an empty advertising every 2.5 seconds and the ones with temperature data?
Thanks for your time

pvvx, 2021-01-31:
> Does this mean that bluetooth is emitting 'advertising bip' every 2.5 second even if there is no connection (bluetooth icon off)?

Yes. The original firmware has an advertising interval of ~ 1.75 seconds, taking readings from the sensor every 4 intervals.
Analog:  https://pvvx.github.io/MHO_C401/img/PowerWrokOriginal.gif

> I'm asking thiy question because original firmware sends data every 10 mins, so I was thinking the purpose was to save battery, but if bluetooth is constantly emitting, I don't understand why it does not send temperature data too

The original firmware sends data every 10 minutes by accelerated transmission of a burst of advertisements with an interval of one hundred microseconds

> What's the difference between an empty advertising every 2.5 seconds and the ones with temperature data?

The question is not for me. To Xiaomi :)

nchieffo, 2021-02-01:
Well also you propose to make measurements less frequently than advertising, so I was just wondering: when you do a measurement you consume lots of energy?
Also, if you don't do measurements the LCD itself will not be upgraded with correct temperature, right?

could you also please clarify some options because I don't find it in the documentation:
- Sensor in "LowPower mode"
- Tx measures
- RF TX Power (I would like to lower this to save battery, but I don't know how to obtain a good value. should I just try setting the lowest value and than check if I receive the packets in my server that consumes the advertisements?)

Thanks

pvvx, 2021-02-01:
There are a lot of options and they all differ when calculating energy consumption.

You can Convert to Percentage of Full Chart yourself:
In the last diagram, this is the maximum duration of an Advertisement with readings of battery voltage and temperature sensor with humidity.
[image]
The tail also has:

I2C SHTV3 sleep command.
I2C refreshes the LCD.

pvvx, 2021-02-01:
> * Sensor in "LowPower mode"

Datasheet Humidity Sensor SHTC3
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/2_Humidity_Sensors/Datasheets/Sensirion_Humidity_Sensors_SHTC3_Datasheet.pdf

pvvx, 2021-02-01:
The CR2032 charges like a battery when connected to a 3.3V source.

nchieffo, 2021-02-03:
Hey Victor,
Unfortunately your explanation is a bit too technical for me, I'm very sorry.
Could you please tell me some basic information about low power? Like: is it precise exactly as normal or not? Do you suggest to enable it?
Thanks again

mvadu, 2021-03-08:
> There are a lot of options and they all differ when calculating energy consumption.
> 
> You can Convert to Percentage of Full Chart yourself:
> In the last diagram, this is the maximum duration of an Advertisement with readings of battery voltage and temperature sensor with humidity.
> [image]
> Battery measurement takes place during operation of the SHTC3 sensor.
> The tail also has:
> 
> I2C SHTV3 sleep command.
> I2C refreshes the LCD.
> 
> Deep-sleep always = 6..8 uA
> 
> ### If sensor LowPower mode:
> If we take a window of 10 ms, TX-RF 0 db, then the average consumption is:
> Advertising, no other action ........ 3.2 mA
> Advertising + sensor reading and update display ........ 4 mA
> 
> ### If sensor Normal mode:
> If we take a window of 18 ms, TX-RF +3 db, then the average consumption is:
> Advertising, no other action ........ 1.85 mA
> Advertising + sensor reading and update display ........ 2.85 mA

Isn't it the other way around? in your comment Low Power mode seems to draw more (3.2 - 4 mA) power than normal mode (1.8-2.8mA)??

jaggil, 2021-03-12:
@mvadu i think the same as you

pvvx, 2021-03-26:
> @mvadu i think the same as you

https://en.wikipedia.org/wiki/Energy
https://en.wikipedia.org/wiki/Kilowatt-hour

10 ms * 3.2 mA = 32 ms*mA
10 ms * 4 mA = 40 ms*mA

18 ms * 1.85 mA = 33.3 ms*mA
18 ms * 2.85 mA = 51.3 ms*mA

closed 2022-05-01 by pvvx