pvvx/ATC_MiThermometer #38: Different ADC values for advertisements

MartMet, 2021-01-31:
Hi,

thanks for this great project. I really appreciate your work. 

I use the LYWSD03MMC with firmware 1.8 and advertising typ "Custom". 

I try to power it from a very small solar panel in combination with a super cap. So far it works reliable down to ~2V. I expierenced that the ADC values from the advertisments and the ADC values from connection mode differ.  For example:

Connection Mode (via TeLinkMiFlasher.html): 2296mV (this one seems close to the real value)
Advert. Mode:        2058mV

Is there any difference in the ADC measurement for this two modes?

I also noticed (running with CR2032 battery) that in advertising mode the battery level indicator byte is sometimes over 100.

pvvx, 2021-01-31:
> I also noticed (running with CR2032 battery) that in advertising mode the battery level indicator byte is sometimes over 100.

?
'''
if(battery_level>100)battery_level=100;
'''
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/battery.c#L82-L83

> Connection Mode (via TeLinkMiFlasher.html): 2296mV (this one seems close to the real value)
> Advert. Mode: 2058mV

The device operates in pulse mode.
When connected, a pulse with one TX-RF - transmission is carried out via one channel.
With advertising uses of 3 pulses TX-RF for 3 channels. More costs.
If the internal resistance of the voltage source is large, then during the measurement, the voltage source sags.

pvvx, 2021-01-31:
> I try to power it from a very small solar panel in combination with a super cap. So far it works reliable down to ~2V. I expierenced that the ADC values from the advertisments and the ADC values from connection mode differ. 

LCD contrast drops dramatically below 2.5 V.
CJMCU-2557 BQ25570 ?
[image]

MartMet, 2021-01-31:
Thanks for your fast reply.
> ?
> '''
> if(battery_level>100)battery_level=100;
> '''
> https://github.com/pvvx/ATC_MiThermometer/blob/master/src/battery.c#L82-L83

Yes, thank you, ive already seen that.

`    Device (new): a4:c1:38:93:22:66 (public), -81 dBm 
	Complete Local Name: 'ATC_932266'
	16b Service Data: <1a1866229338c1a4c3068015770b51340d>`
--> 1a1866229338c1a4c3068015**77**0b51340d          So 0x77 should be the battery indicator, or am i addressing the wrong byte?

> The device operates in pulse mode.
> When connected, a pulse with one TX-RF - transmission is carried out via one channel.
> With advertising uses of 3 pulses TX-RF for 3 channels. More costs.
> If the [...]

Ah ok i see. Would it help to delay the ADC Measurement?

MartMet, 2021-01-31:
> LCD contrast drops dramatically below 2.5 V.
> CJMCU-2557 BQ25570 ?
> [image]

Cool converter. :) Actually i still can read the display, what matters most to me is, that BLE is still running.

MartMet, 2021-01-31:
> --> 1a1866229338c1a4c3068015**77**0b51340d So 0x77 should be the battery indicator, or am i addressing the wrong byte?
In fact i was addressing the wrong byte, the correct byte should be 0x51. And the voltage bytes are 0x0b77. 
(As the values looked reasonable i even was wondering why the voltage endianess was reversed :D)

Thanks for your fast response. :) I created a PR to fix the doc.

closed 2021-01-31 by MartMet