pvvx/ATC_MiThermometer #41: [Feature request] Support for CGG1 possible?

MacSass, 2021-02-03:
Hello, great firmware / solution you have build. Works perfectly - love it on my LYWSD03MMC.
Incredible what is possible and how a commercial product can be made so much better! Thank you!

I´m wondering if it was possible to add the CGG1 model. It seems to use the same type of encrypted advertisement in recent FW, but has a very nice e-paper display with low energy consumption. 
Link to respective model here: [www.aliexpress.com/item/1000008051178.html](https://www.aliexpress.com/item/1000008051178.html)

Would love to see that supported. Thanks again for making and documenting all this for us.

Regards - MacSass

pvvx, 2021-02-03:
Hi
There were like 2 models in round cases? And there is already enough trash ... :)

atc1441, 2021-02-03:
@MacSass as you kind of see in these FCC pictures https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168 the CGG1 uses and nRF52832 as SOC, that would mean a custom rom is a complete new project.
This also counts for other versions of the Xiaomi devices.

One other device i know that has an tlsr is the mosquito repellend

pvvx, 2021-02-03:
Various, from Xiaomi, with temperature sensors, etc:

LYWSD02MMC https://ixbt.online/live/images/original/00/94/02/2020/07/10/8cfc21dc51.jpg SOC Dialog DA14585
WSDCGQ01LM 
LYWSDCGQ/01ZM https://blog.uaid.net.ua/wp-content/uploads/2020/12/MiSmart_temphum_01.jpg https://blog.uaid.net.ua/ble-temperature-sensor/#more-4386
 JN5169
https://faire-ca-soi-meme.fr/domotique/2017/02/27/hack-xiaomi-mi-smart-temperature-and-humidity-sensor/
https://i2.wp.com/faire-ca-soi-meme.fr/wp-content/uploads/2017/05/IMG_20170528_115624-e1495982583660.jpg

EFR32MG13
https://i1.wp.com/faire-ca-soi-meme.fr/wp-content/uploads/2019/07/capteur_temp_humidity_demontage.jpg
https://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home

pvvx, 2021-02-03:
> .. that would mean a custom rom is a complete new project.

Everything has an SDK, datasheet, ...
If 2 people take on each type, then in a month everything from Xiaomi will be with non-standard firmware :)

atc1441, 2021-02-03:
@pvvx Thats true.

Only some ota update method could be harder to crack, but hardware update should work on all of them

pvvx, 2021-02-03:
Of the BLE, Xiaomi will only have these chip manufacturers:

Nordic | 51 Series | https://github.com/MiEcosystem/mijia_ble_standard/tree/nordic_legacy
-- | -- | --
Nordic | 52 Series | https://github.com/MiEcosystem/mijia_ble_standard/tree/nordic
Silicon Labs | BG13 | https://github.com/MiEcosystem/mijia_ble_standard/tree/silabs
Telink | TLSR825x | https://github.com/MiEcosystem/mijia_ble_standard/tree/telink
Realtek | RTL8752x / RTL8762x | https://github.com/MiEcosystem/mijia_ble_standard/tree/realtek
Dialog | DA145xx | https://github.com/MiEcosystem/mijia_ble_standard/tree/Dialog
Dialog | DA146xx | https://github.com/MiEcosystem/mijia_ble_standard/tree/da146xx

For TLSR825x from Telink https://yadi.sk/d/g5fV7WD1EaUdRQ

All have the same OTA connection coding.

atc1441, 2021-02-03:
"All have the same OTA connection coding."

The tlsr uses the Telink ota connection coding so i doubt it?!

pvvx, 2021-02-03:
They all use their standard options. But binding and bindkey are similar.
So there is also OTA from mi.

pvvx, 2021-02-03:
The advertising format is also described in detail:
https://iot.mi.com/new/doc/embedded-development/ble/ble-mibeacon
BLE MiBeacon protocol:
The broadcast message with the following information in the broadcast identifier is a broadcast conforming to the MiBeacon protocol:

"Service Data" (0x16) in advertising contains Mi Service (UUID: 0xFE95)
The "Manufacturer Specific Data" (0xFF) in the scan response contains the Xiaomi company identification code (ID: 0x038F)
Note: "advertising" or "scan response" are defined in a unified format.

Field | Name | Types of | Length (byte) | Mandatory (M) / Optional (O) | Description
-- | -- | -- | -- | -- | --
1 | Frame Control | Bitmap | 2 | M | Control bit, see the specific definition in the table below
2 | Product ID | U16 | 2 | M | Product ID, unique for each type of product, pid need millet IoT Developer Platform applications
3 | Frame Counter | U8 | 1 | M | Serial number, used for de-duplication, different event or attribute reporting requires different Frame Counter
4 | MAC Address | U8 | 6 | C.1 | Device Mac address
5 | Capability | U8 | 1 | C.1 | Equipment capacity, see the following table definition for details
6 | I/O capability | U8 | 2 | C.2 | I/O capability. This field is currently only used for high-security BLE access, and only MiBeacon v5 is supported. It is only used before binding; when the binding is completed and an event is reported (such as door opening, door closing), this field is no longer needed. See the definition in the table below for details
7 | Object | U8 | n (according to actual needs) | C.1 | Trigger events or broadcast properties, see Mijia BLE Object protocol for details
8 | Random Number | U8 | 3 | C.1 | If encrypted, it is a required field, combined with Frame Counter to become a 4-byte Counter for anti-replay
9 | Message Integrity Check | U8 | 4 | C.1 | If encrypted, it is a required field, MIC four bytes

and etc...

MacSass, 2021-02-03:
Ok guys,
I understand it is not so easy to integrate as I was hoping ... thanks for your feedback.
Xiaomi really has a huge portfolio of "smarthome" devices build up ...
Anyhow - I´ll order some more LYWSD03MMC if needed - they are cheap enough ..

Thanks - I´ll watch this to see if you cracsk come up with anything new, unfortunately I don´t consider my coding know-how advanced enough to attack those other devices ...

MacSass

pvvx, 2021-02-03:
> Anyhow - I´ll order some more LYWSD03MMC if needed - they are cheap enough ..

MHO-C401 more beautiful.
The rest - as it becomes available and in mood...
I'm waiting for Xiaomi to have PHY62x2. The nRF, EFR, Dialog, STM, Silicon Labs are boring to me.

kelchm, 2021-02-10:
If anyone is interested in taking a look, I have some very early stages efforts for the CGG1 I'm documenting [here](https://github.com/kelchm/cgg1-thermometer-firmware/wiki). It hasn't made it to the wiki yet, but I have managed to at least [dump the stock firmware](https://github.com/kelchm/cgg1-thermometer-firmware/issues/1).

PS: Thanks @pvvx for your work on the LYWSD03MMC 💯

MacSass, 2021-02-10:
Hi @kelchm ,
thanks for the update - that is interesting progress.
From you pictures I´m really hoping it will be able to OTA those devices too - dismanteling seems to be a hard job.
Unfortunately you guys are way more advanced than me, so I´m of no real help, but for sure I will be watching your progress :-)
Thanks for the work and update ...
Regards - MacSass

pvvx, 2021-02-23:
> TLSR использует кодировку соединения Telink ota, поэтому я в этом сомневаюсь ?!

https://github.com/scooterhacking/mijia_ble_libs/tree/master/gatt_dfu

pvvx, 2021-03-12:
CCG1 "Qingping Temp & RH Monitor"

---

BT Address: 58:2d:34:11:c3:b2
Number of Services: 10
BT 4.2 Secure Connection: False
Device Connected: True
Service Name: GenericAccess
Service UUID: 00001800-0000-1000-8000-00805f9b34fb
Characteristic Name: DeviceName - User Description: - Handle: 2 - Value: Qingping Temp & RH M
Characteristic Name: Appearance - User Description: - Handle: 4 - Value: 00-00
Characteristic Name: PeripheralPreferredConnectionParameters - User Description: - Handle: 6 - Value: 14-00-28-00-00-00-E8-03
Service Name: GenericAttribute
Service UUID: 00001801-0000-1000-8000-00805f9b34fb
Characteristic Name: ServiceChanged - User Description: - Handle: 9 - Value:
Service Name: Battery
Service UUID: 00001 80f-0000-1000-8000-00805f9b34fb
Characteristic Name: BatteryLevel - User Description: - Handle: 13 - Value: 64
Service Name: 0001 0203-0405-0607-0809-OaObOcOdl 912
Service UUID: 0001 0203-0405-0607-0809-OaObOcOdl 912
Characteristic Name: 00010203-0405-0607-0809-OaObOcOd2bl2 - User Description: OTA - Handle: 17 - Value: 00
Service Name: 00010203 -0405-0607-0809-OaObOcOd2b2O
Service UUID: 00010203-0405-0607-0809-OaObOcOd2b2O
Characteristic Name: 00010203-0405-0607-0809-OaObOcOd2b2l - User Description: Vendor - Handle: 21 - Value: 00
Service Name: 65173
Service UUID: 0000fe95-0000-1000-8000-00805f9b34fb
Characteristic Name: 4 - User Description: Version - Handle: 26 - Value: 1.0.1_0093
Characteristic Name: 16 - User Description: Authentication - Handle: 29 - Value:
Characteristic Name: 23 - User Description: ota_ctrl - Handle: 33 - Value:
Characteristic Name: 24 - User Description: ota_data - Handle: 37 - Value:
Characteristic Name: 25 - User Description: standard - Handle: 41 - Value:
Service Name: 00000100-0065-6c62-2e74-6f696d2e696d
Service UUID: 00000100-0065-6c62-2e74-6f696d2e696d
Characteristic Name: 00000101-0065-6c62-2e74-6f696d2e696d - User Description: STDIO_RX - Handle: 46 - Value:
Characteristic Name: 00000102-0065-6c62-2e74-6f696d2e696d - User Description: STDIO_TX - Handle: 50 - Value:
Service Name: 22210000-554a-4546-5542-46534450464d
Service UUID: 22210000-554a-4546-5542-46534450464d
Characteristic Name: 1 - User Description: - Handle: 55 - Value:
Characteristic Name: 2 - User Description: - Handle: 57 - Value:
Characteristic Name: 3 - User Description: - Handle: 60 - Value:
Characteristic Name: 4 - User Description: - Handle: 62 - Value:
Characteristic Name: 256 - User Description: - Handle: 65 - Value:
Service Name: Devicelnformation
Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
Characteristic Name: ManufacturerNameString - User Description: - Handle: 69 - Value: Qingping Technology (Beijing) Co., Ltd.
Characteristic Name: ModelNumberString - User Description: - Handle: 71 - Value: CGG1
Characteristic Name: SerialNumberString - User Description: - Handle: 73 - Value: 1234
Characteristic Name: HardwareRevisionString - User Description: - Handle: 75 - Value: 0001
Characteristic Name: FirmwareRevisionString - User Description: - Handle: 77 - Value: 1.0.1_0093
Service Name: 6e400001-b5a3-f393-eOa9-e50e24dcca9e
Service UUID: 6e400001-b5a3-f393-eOa9-e50e24dcca9e
Characteristic Name: 6e400002-b5a3-f393-eOa9-e50e24dcca9e - User Description: - Handle: 80 - Value:
Characteristic Name: 6e400003-b5a3-f393-eOa9-e50e24dcca9e - User Description: - Handle: 82 - Value:

---

https://pvvx.github.io/CGG1/OriginalPower.html

kelchm, 2021-03-13:
> https://pvvx.github.io/CGG1/

Very interesting! Can't say I expected there to be a newer revision of the hardware that is using the TLSR8253 instead of the nRF52832.

pvvx, 2021-03-13:
> Very interesting! I never would have expected there to be a newer revision of the CGG1 hardware that is using the TLSR8253 instead of the nRF52832.

Nordic is a Norwegian company listed on the Oslo stock exchange.
Telink - China.
Mijia - China.
Xiaomi - China.
nRF chip price > TLSR

pvvx, 2021-03-13:
https://www.qingping.co/temp-rh-monitor/overview
```
One Button Cell Lasts 8 ~ 12 Months
* This product uses CR2430 battery, it normally lasts 8 ~ 12 months in indoor environments (data provided by Qingping lab). The actual standby time may differ depending on the environment and battery.
```
All batteries in the kits came discharged (less than 0.2 V).
Russian Winter. :)

MacSass, 2021-03-13:
> 
> 
> > https://pvvx.github.io/CGG1/
> 
> Very interesting! I never would have expected there to be a newer revision of the CGG1 hardware that is using the TLSR8253 instead of the nRF52832.

Hi, unfortunately I don´t feel like I´m able to understand everything you guys are talking about ... so I have two questions:

- Is there an easy way to see if CGG1 is of the TLSR8253 type or the older type?
- Does using TLSR8253 make it easier for you guys to develop a custom firmware?

Regards - MacSass

pvvx, 2021-03-13:
> * Is there an easy way to see if CGG1 is of the TLSR8253 type or the older type?

Round thermometers from Qingping are of two types - "H" and "M".
https://www.qingping.co/temp-rh-monitor/overview

Marking on the box -> https://pvvx.github.io/CGG1 

> * Does using TLSR8253 make it easier for you guys to develop a custom firmware?

As soon as there is free time, so will the firmware immediately. It is necessary to correct the procedures for working with E-inc.

MacSass, 2021-03-13:
> > * Is there an easy way to see if CGG1 is of the TLSR8253 type or the older type?
> 
> Round thermometers from Qingping are of two types - "H" and "M".
> https://www.qingping.co/temp-rh-monitor/overview
> 
> Marking on the box -> https://pvvx.github.io/CGG1
> 
> https://aliexpress.com/item/1005001914179317.html
> 
So, from your links, is my understanding right:
- "H" is the Homekit version and uses nRF chip, needs completely new reverse engineering
- "M" is the Mi-Home version and uses TLSR8253 chipset. A TLSR8253 custom firmware would be easier for you, but requires some modifications due to e-ink display.

That would be great news ...

PS: You Ali-Express link goes to a LCD version, which again is something totally different from my point of view. I have an e-ink one and the display is way better compared to LCD ...

pvvx, 2021-03-13:
I ordered it there:
https://aliexpress.ru/item/1005001914179317.html
"Xiaomi Bluetooth Temperature and Humidity Sensor Lite Version Data Storage **Ink** LCD Screen Thermometer Support Mi Home APP"

pvvx, 2021-03-13:
> * "H" is the Homekit version and uses nRF chip, needs completely new reverse engineering

Unknown - These are Apple and HomeKit, which I do not support.

> * "M" is the Mi-Home version and uses TLSR8253 chipset. A TLSR8253 custom firmware would be easier for you, but requires 

The firmware from Xiaomi LYWSD03MMC will work, but there will be no indication on the display.

MacSass, 2021-03-13:
> I ordered it there:
> https://aliexpress.ru/item/1005001914179317.html
> "Xiaomi Bluetooth Temperature and Humidity Sensor Lite Version Data Storage **Ink** LCD Screen Thermometer Support Mi 

Home APP"
Did you already receive it? Because the pictures show the LCD version (visible by the small darker grey area at the bottom, which the e-ink version does not have). 
Also the descriptions says "**Ink LCD screen**" and the chinese often don´t take wording so serious, so unless you received it with e-ink already, I would guess this version is the LCD one, because it is also unusual cheap for a real e-ink version.

I ordered this one - with "real" e-ink:
[https://www.aliexpress.com/item/32990058321.html](url)
as you can see the picture looks different, no grey area at the bottom of the display.

I´ll hope you guys will be able to support that in the future, including the nice e-ink display ...

pvvx, 2021-03-13:
[image]
All photos of one of them are here: https://pvvx.github.io/CGG1

pvvx, 2021-03-13:
> because it is also unusual cheap for a real e-ink version.

https://cleargrass.world.tmall.com/
[image]
[image]

pvvx, 2021-03-13:
> From you pictures I´m really hoping it will be able to OTA those devices too - dismanteling seems to be a hard job.

There are two OTA functions on the device. From Telink and from Mijia.

pvvx, 2021-03-14:
EPD Segments:
https://pvvx.github.io/CGG1/img/EPD_segments.gif
https://pvvx.github.io/CGG1

kelchm, 2021-03-17:
> So, from your links, is my understanding right:
>
> - "H" is the Homekit version and uses nRF chip, needs completely new reverse engineering
>
> - "M" is the Mi-Home version and uses TLSR8253 chipset. A [...]

Unfortunately I don't think it's that simple -- I purchased the M variant in the US and received the nRF variant.

I'm guessing that you're much more likely to get the TLSR variant ordering directly from China, but probably still a bit of a game of chance.

kelchm, 2021-03-17:
> I ordered it there:
> https://aliexpress.ru/item/1005001914179317.html
> "Xiaomi Bluetooth Temperature and Humidity Sensor Lite Version Data Storage **Ink** LCD Screen Thermometer Support Mi Home APP"

Strange -- like @MacSass I only see the 'Lite' version in this listing with the LCD screen:

[image]

@pvvx Can you check if the seller updated the listing after you placed your order?

For what it's worth, I've rolled the dice with ordering two more CGG1 (M) from two different vendors on AliExpress:
1. https://www.aliexpress.com/item/1005001552641195.html
2. https://www.aliexpress.com/item/1005002204565339.html

I will report back once I receive them to let everyone know if either ended up being being the newer hardware revision with the TLSR8253.

MacSass, 2021-03-17:
Thanks! Looking forward to hear what you actually receive ...

pvvx, 2021-03-17:
I ordered without understanding what was there - purely for the collection. I didn't care - I was looking for the TLSR825x chips.
The price was 912.29 rubles, and now it is already 1,498.32 rubles.
[image]
[image]

pvvx, 2021-03-17:
Difference in the image of E-Ink and LCD:
[image]

toomyem, 2021-03-18:
Hi guys. What about "Lite" version? Any chance of hacking this one?
https://www.qingping.co/temp-rh-monitor-lite/specifications

pvvx, 2021-03-18:
> Привет, народ. А как насчет "Lite" версии? Есть ли шанс взломать это?
> https://www.qingping.co/temp-rh-monitor-lite/specifications

Сheck the UUID information in nRF Connect. If there is an OTA from Telink then - disassemble the device, take a photo, write what chips are there...  Publish photos and data... Then I will order these thermometers, when they come I will write the firmware ... :)

pvvx, 2021-03-18:
Another **CGG1** - 2018.11. **Not TLSR825x.**
[image]
[image]
[image]

pvvx, 2021-03-21:
There are more than 3 variants of the CGG1 on sale.
**CGG1**
https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168
**CGG1H**
https://fccid.io/2AQ3F-CGG1H/Internal-Photos/Internal-Photos-4723738
**CGG1-M**
https://pvvx.github.io/CGG1

Who will find more variants of CGG1? :)

chunyianliew, 2021-03-25:
I received the following overview directly from QingPing after I contacted them complaining about all the different versions of the CGG1 variants I received from one shop (multiple orders, same variant within one order):

Picture | Product Name on Package | Display Type | Bluetooth Name | Works with Mi Home | Current Status | Website
-- | -- | -- | -- | -- | -- | --
​[image] | ClearGrass Temp & RH Monitor | E Ink | "Goose" or "ClearGrass Temp & RH" (after update, the name changes) | No | Discontinued | N/A
​​[image] | Qingping Temp & RH Monitor M Version | E Ink | Qingping Temp RH M | Yes | Normal | https://www.qingping.co/temp-rh-monitor/overview
​​[image]| Qingping Temp & RH Monitor Lite | Segment LCD | Qingping Temp RH Lite | Yes | Normal | https://www.qingping.co/temp-rh-monitor-lite/overview

After trying to integrate these monitors with Home Assistant using ESPHome I discovered the following about the three different E-Ink variants I currently possess:

1. "ClearGrass Temp & RH" - firmware version 1.1.2_0036, easy to integrate, no bluetooth encryption bindkey required
2. "Goose" -> firmware version 1.5, seems to require an bluetooth encryption bindkey, not able to retrieve this bindkey yet.
3. "QingPing Temp RH M" - firmware version 1.0.1_0093, requires bluetooth encryption bindkey for integration, managed to retrieve the bindkey following the instructions provided here: https://sequr.be/blog/2021/01/home-automation-room-temperature-monitoring-using-xiaomi-temperature-sensor-esp32-and-esphome/#getting-the-bindkey-of-the-cgg1

pvvx, 2021-03-25:
> 3. "QingPing Temp RH M" - firmware version 1.0.1_0093, requires bluetooth encryption bindkey for integration, managed to retrieve the bindkey following the instructions provided here: [...]

The old Bindkey is read in custom firmware.  The old BindKey is read in the custom firmware. BindKey recovery also works for flashing to original firmware...
All devices with OTA have the ability to read keys and restore firmware back. In one pass. For this, a special OTA version is possible...

mattkelch-wk, 2021-03-31:
I have some good news -- I received the first of the CGG1's that I ordered from China and it turned out to be the newer Telink based hardware revision that @pvvx also reported receiving. While I obviously can't guarantee anything on what you'll actually receive, [this](https://www.aliexpress.com/item/1005002204565339.html) is the listing that I ordered from. 

[image]

I've successfully flashed the v2.9 release and all seems to be working as expected. Thanks again to @pvvx for going out on a limb to order some CGG1s, discovering this new hardware revision and adding support for it in the project. 💯

kelchm, 2021-03-31:
I have some good news -- I received the first of the CGG1's that I ordered from China and it turned out to be the newer Telink based hardware revision that @pvvx also reported receiving. While I obviously can't guarantee anything on what you'll actually receive, [this](https://www.aliexpress.com/item/1005002204565339.html?spm=a2g0s.9042311.0.0.24a24c4dwprpUd) is the listing that I ordered from.

[image]

I've successfully flashed the v2.9 release and all seems to be working as expected. Thanks again to @pvvx for going out on a limb to order some CGG1s, discovering this new hardware revision and adding support for it in the project. 💯

toomyem, 2021-03-31:
How did you flash this? I have the same version as yours. I used telink flasher page, but after clicking "Do activate", log says "activating, please wait..." and nothing happens (I waited a few minutes). I also tried TelinkOTA, but after clicking "Start Flashing", I got "Update error: some error while sending char data". I have used CGG1_v29.bin firmware.

simoncheeseman, 2021-03-31:
I've just received a new CGG1 (M Version) and I have the same problem - the flasher is stuck on "Activating, please wait...". I've tried flashing from a PC and an Android phone (Chrome on both). I was able to successfully flash an LYWSD03MMC a few days earlier.

pvvx, 2021-03-31:
>  I also tried TelinkOTA, but after clicking "Start Flashing"

TelinkOTA not work in CGG1-M. This is a program for OTA projects with Telink SDK, no third-party (mijia) protections.

pvvx, 2021-04-03:
CGG1 variants produced by [Qingping.co](https://www.qingping.co/temp-rh-monitor/overview) ([ClearGrass](https://www.cleargrass.com/))
---

**CGG1-old** ver1, Confirmed FCC ID: [2AQ3F-CGG1](https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168)
https://pvvx.github.io/CGG1_old
nRF52810 Chipset. It has a small Flash in the nRF52810 chip, an additional SPI-Flash is used for OTA - Limited in functionality. Terrible economy on details. Lack of any recommended elements for stable operation of nRF52 chips from battery.
Production 2018
Increased cost because a chip from Nordic is used.

**CGG1-?** ver2, Unknown FCC ID.
https://github.com/kelchm/cgg1-thermometer-firmware
nRF52832 Chipset. Terrible economy on details. Lack of any recommended elements for stable operation of nRF52 chips from battery. 
Production 2018..2019 ?
Increased cost because a chip from Nordic is used and Mijia.

**CGG1-M** ver3,  has an official fake FCC ID: 2AQ3F-CGG1
https://pvvx.github.io/CGG1
TLSR8253 Chipset. The workmanship is better than previous models. Additional circuitry of additional elements for better and long-term stability from battery.
Production 2020+
The cost is much lower, because the chip used is not Nordic, but the Chinese Telink.

**CGG1-H** ver4, Confirmed FCC ID: [2AQ3F-CGG1H](https://fccid.io/2AQ3F-CGG1H/Internal-Photos/Internal-Photos-4723738)
Unknown Chipset
Production 2020+
Increased cost due to the use of Apple HomeKit.

---

Alternative firmware is only possible for **CGG1-M**.
The first version of **CGG1-old** is in development. Will not be in this repository.

MacSass, 2021-04-03:
Hi, is there any way for me to identify which version I have, without opening it, if I do not have the packaging anymore?

Are there outer differences or can I identify it by BT connection?

pvvx, 2021-04-03:
> Are there outer differences or can I identify it by BT connection?

There are no external differences. Absolutely.
It is perfectly possible if you poke the camera through the 2 mm battery compartment and look at the details of the board. You can use an x-ray.

MacSass, 2021-04-03:
Great - thank you, my X-Ray is currently broken :-) ... but I´ll try the nRF connect option.

Regards - MacSass

MacSass, 2021-04-03:
Ok, if I´m not mistaken, I would take it this is a CGG1-M with Telink, right? 

[image]

Looks like I was lucky? That one would be supported by alternative firmware with OTA flash, and the e-paper display would continue to work, right?

kelchm, 2021-04-03:
> Looks like I was lucky? That one would be supported by alternative firmware with OTA flash, and the e-paper display would continue to work, right?

Yep -- you should be good to go to flash with the [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html). 👍 

It look like it's turning out to be quite easy to pick up these Telink based versions. The second one I ordered from China arrived today with a box dated `2021.01`, and as expected is indeed the Telink variant. For this one, I ordered the *For Mijia App* option from [this listing](https://www.aliexpress.com/item/1005001552641195.html) -- obviously YMMV for anyone that chooses to also order from here.

[image]

pvvx, 2021-04-03:
> It look like it's turning out to be quite easy to pick up these Telink based versions. The second one I ordered from China arrived today with a box dated `2021.01`, and as expected is indeed the [...]

It looks like "2020 new H version" is also on Telink.
[image]

Need firmware for OTA tests...

kroon040, 2021-04-27:
> > I also tried TelinkOTA, but after clicking "Start Flashing"
> 
> TelinkOTA not work in CGG1-M. This is a program for OTA projects with Telink SDK, no third-party (mijia) protections.
> 
> The [atc1441 TelinkFlasher](https://atc1441.github.io/TelinkFlasher.html) will not work with the CGG1-M (and many others). The data is not signed correctly and when connecting it expects a UUID that CGG1-M does not have.
> The correct signature for the binary OTA firmware file is put by the utility from the Telink SDK, and not from the Ai-Thinker-Open Telink_825X_SDK (!): https://github.com/pvvx/ATC_MiThermometer/tree/master/utils
> All original OTA files (Xiaomi LYWSD03MMC, MHO-C401, CGG1-M, ...) are signed with this particular utility.
> There is no such thing for Linux.
> The size of the binary file must be correctly specified in the bootloader header ([and have 0xXXXX4](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/cstartup_825x.S#L76)).
> 
> Use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html and act consistently...

What is the correct way of flashing? I found the CGG1_v31.bin file but how , it doesn't work with  activate.

pvvx, 2021-04-27:
> > Use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html and act consistently...
> 
> What is the correct way of flashing? I found the CGG1_v31.bin file but how , it doesn't work with activate.

All Ok.
To activate, press the button before connecting. Described in the standard instructions for CGG1-M.
```
00:42:05: Searching for devices
00:42:10: Connecting to: Qingping Temp & RH M
00:42:15: Detected Telink OTA service
00:42:15: Connected
00:42:18: Activating now, please wait...
00:42:20: Activation successfull
00:42:21: Received device infos are correct
00:42:21: Login successfull
GraphOriginal
00:42:24: Load firmware file 'CGG1_v31.bin'...
00:42:25: File size: 64148 bytes
00:42:25: Count: 4010
00:42:27: Start DFU
00:42:52: Update done after 24.112 seconds
00:42:56: Disconnected.
```

kroon040, 2021-04-28:
> All Ok.
> To activate, press the button before connecting. Described in the standard instructions for CGG1-M.
>
> ```
> 00:42:05: Searching for devices
> 00:42:10: Connecting to: Qingping Temp & RH M [...]

Tnnx, I got it working.

sermayoral, 2021-04-29:
Hi everybody.

I has the ClearGrass versión (UUID 00010203-0405-0607-0809-0a0b0c0d1912) that it doesn't work: https://github.com/pvvx/ATC_MiThermometer/issues/81

Waiting for a hope to find a solution to hack this model :-)

yerkanian, 2021-05-01:
Just to clarify the activation/flashing procedure for the new CGG1-M (with GOOSEM20XXXXXXX 106308/00096720 or similar markings and production date 2020.03) once again:

Press and hold the button on the back of CGG1-M for 2 seconds until the Bluetooth icon starts blinking on the e-ink display, and then click the 'Connect' and 'Do Activation' buttons on the TeLink Flasher. Took me an hour to figure out why the activation was timing out...

Once again, great job on the firmware, @pvvx!

kelchm, 2021-05-13:
FYI -- it looks like the new CGG1 'M' is now [available from Amazon](https://www.amazon.com/dp/B08N6BL434) in the US. I ordered one the other day and while not as new as those I've ordered directly from China, its still new enough to be the telink variant.

przemhb, 2021-11-10:
I've just received "ClearGrass Temp & RH Monitor **M** Version", production date 2020.01. It came with FW version 1.1.2_0020. 
TelinkMiFlasher returns an error "NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device." nRF Connect among UUIDs shows 0xFE59. 

Correct me if I am wrong, but it seems it is yet another variant - marked as CGG1-M, but in fact an old CGG1.

pvvx, 2021-11-11:
>  nRF Connect among UUIDs shows 0xFE59.

UUID from Nordic DFU - 0xFE59
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fgroup__nrf__dfu__ble.html

closed 2022-05-01 by pvvx

Stephan3, 2022-07-15:
Hey i bought a few devices within the last two years from different Sources and got lywsdcgq/01zm everytime. I opened one to share pictures here. I like the idea this project broadcasts the data.

[image]
[image]
[image]
[image]
[image]

I havent found the information here if its supported allready. the webflasher had not connected to it. I have about 10 here ready to brick.

[2 more replies]