pvvx/ATC_MiThermometer #44: Incorrect negative temperature values retrieved using GetMemo

skaldo, 2021-02-12:
Hi @pvvx,
thank you for this great fork of the Aaron's work!

I was playing around and found out that the GetMemo returns incorrect values for negative temperatures, see log below.

Temperature received by normal advertising looks fine (-4°C), but temperature from memory is at about 171°C. One would almost say, that simply adding the magic constant 175 from the datasheet of Sensirion would solve this 😄

Any ideas what could be the cause?

<details>
  <summary>Logs</summary>

```
Send cmd ok
TelinkMiFlasher.html:810 Vbat: 2546 mV , Temp: -4.12°C, Humi: 56.27%, Count: 53, flg: 22
TelinkMiFlasher.html:810 Send cmd GetMemo: (3532000000)...
TelinkMiFlasher.html:814 Send cmd GetMemo ok
TelinkMiFlasher.html:814 2021-02-12 21:44:28 # Vbat: 2541 mV , Temp: 170.67°C, Humi: 56.71%, Count: 1
TelinkMiFlasher.html:814 2021-02-12 21:34:27 # Vbat: 2542 mV , Temp: 170.78°C, Humi: 56.72%, Count: 2
TelinkMiFlasher.html:814 2021-02-12 21:24:26 # Vbat: 2542 mV , Temp: 170.82°C, Humi: 56.62%, Count: 3
TelinkMiFlasher.html:814 2021-02-12 21:14:25 # Vbat: 2547 mV , Temp: 170.9°C, Humi: 56.45%, Count: 4
TelinkMiFlasher.html:814 2021-02-12 21:04:24 # Vbat: 2546 mV , Temp: 170.95°C, Humi: 56.34%, Count: 5
TelinkMiFlasher.html:814 2021-02-12 20:54:24 # Vbat: 2544 mV , Temp: 171.03°C, Humi: 56.17%, Count: 6
TelinkMiFlasher.html:814 2021-02-12 20:44:23 # Vbat: 2542 mV , Temp: 171.11°C, Humi: 56.08%, Count: 7
TelinkMiFlasher.html:814 2021-02-12 20:34:23 # Vbat: 2543 mV , Temp: 171.15°C, Humi: 55.91%, Count: 8
TelinkMiFlasher.html:814 2021-02-12 20:24:22 # Vbat: 2545 mV , Temp: 171.22°C, Humi: 55.84%, Count: 9
TelinkMiFlasher.html:814 2021-02-12 20:14:21 # Vbat: 2540 mV , Temp: 171.35°C, Humi: 55.5%, Count: 10
TelinkMiFlasher.html:814 2021-02-12 20:04:20 # Vbat: 2523 mV , Temp: 171.4°C, Humi: 55.15%, Count: 11
TelinkMiFlasher.html:814 2021-02-12 19:54:20 # Vbat: 2537 mV , Temp: 171.46°C, Humi: 54.96%, Count: 12
TelinkMiFlasher.html:814 2021-02-12 19:44:19 # Vbat: 2594 mV , Temp: 171.52°C, Humi: 54.56%, Count: 13
TelinkMiFlasher.html:814 2021-02-12 19:34:19 # Vbat: 2547 mV , Temp: 171.63°C, Humi: 54.13%, Count: 14
TelinkMiFlasher.html:814 2021-02-12 19:24:16 # Vbat: 2543 mV , Temp: 171.75°C, Humi: 53.8%, Count: 15
TelinkMiFlasher.html:814 2021-02-12 19:14:15 # Vbat: 2556 mV , Temp: 171.79°C, Humi: 53.42%, Count: 16
TelinkMiFlasher.html:814 2021-02-12 19:04:15 # Vbat: 2581 mV , Temp: 171.78°C, Humi: 53.1%, Count: 17
TelinkMiFlasher.html:814 2021-02-12 18:54:15 # Vbat: 2592 mV , Temp: 171.83°C, Humi: 52.77%, Count: 18
TelinkMiFlasher.html:814 2021-02-12 18:44:11 # Vbat: 2595 mV , Temp: 171.92°C, Humi: 52.44%, Count: 19
TelinkMiFlasher.html:814 2021-02-12 18:34:07 # Vbat: 2603 mV , Temp: 172.05°C, Humi: 51.9%, Count: 20
TelinkMiFlasher.html:814 2021-02-12 18:24:03 # Vbat: 2602 mV , Temp: 172.23°C, Humi: 51.26%, Count: 21
TelinkMiFlasher.html:814 2021-02-12 18:14:00 # Vbat: 2589 mV , Temp: 172.39°C, Humi: 50.44%, Count: 22
TelinkMiFlasher.html:814 2021-02-12 18:03:56 # Vbat: 2590 mV , Temp: 172.5°C, Humi: 49.76%, Count: 23
TelinkMiFlasher.html:814 2021-02-12 17:53:52 # Vbat: 2572 mV , Temp: 172.66°C, Humi: 48.75%, Count: 24
TelinkMiFlasher.html:814 2021-02-12 17:43:49 # Vbat: 2573 mV , Temp: 172.94°C, Humi: 47.88%, Count: 25
TelinkMiFlasher.html:814 2021-02-12 17:33:45 # Vbat: 2584 mV , Temp: 173.19°C, Humi: 47.21%, Count: 26
TelinkMiFlasher.html:814 2021-02-12 17:23:41 # Vbat: 2578 mV , Temp: 173.47°C, Humi: 46.5%, Count: 27
TelinkMiFlasher.html:814 2021-02-12 17:13:38 # Vbat: 2576 mV , Temp: 173.85°C, Humi: 45.37%, Count: 28
TelinkMiFlasher.html:814 2021-02-12 17:03:34 # Vbat: 2580 mV , Temp: 174.35°C, Humi: 44.41%, Count: 29
TelinkMiFlasher.html:814 2021-02-12 16:53:30 # Vbat: 2589 mV , Temp: 0.32°C, Humi: 43.11%, Count: 30
TelinkMiFlasher.html:814 2021-02-12 16:43:26 # Vbat: 2591 mV , Temp: 0.64°C, Humi: 42.58%, Count: 31
TelinkMiFlasher.html:814 2021-02-12 16:33:22 # Vbat: 2587 mV , Temp: 0.39°C, Humi: 42.43%, Count: 32
TelinkMiFlasher.html:814 2021-02-12 16:23:19 # Vbat: 2586 mV , Temp: 0.36°C, Humi: 42.31%, Count: 33
TelinkMiFlasher.html:814 2021-02-12 16:13:15 # Vbat: 2586 mV , Temp: 0.43°C, Humi: 42.3%, Count: 34
TelinkMiFlasher.html:814 2021-02-12 16:03:11 # Vbat: 2588 mV , Temp: 0.56°C, Humi: 42.47%, Count: 35
TelinkMiFlasher.html:814 2021-02-12 15:53:07 # Vbat: 2591 mV , Temp: 0.86°C, Humi: 42.13%, Count: 36
TelinkMiFlasher.html:814 2021-02-12 15:43:04 # Vbat: 2590 mV , Temp: 0.9°C, Humi: 41.82%, Count: 37
TelinkMiFlasher.html:814 2021-02-12 15:33:00 # Vbat: 2590 mV , Temp: 0.9°C, Humi: 40.58%, Count: 38
TelinkMiFlasher.html:814 2021-02-12 15:22:56 # Vbat: 2592 mV , Temp: 0.98°C, Humi: 40.6%, Count: 39
TelinkMiFlasher.html:814 2021-02-12 15:12:53 # Vbat: 2594 mV , Temp: 1.19°C, Humi: 39.57%, Count: 40
TelinkMiFlasher.html:814 2021-02-12 15:02:49 # Vbat: 2596 mV , Temp: 1.43°C, Humi: 40.38%, Count: 41
TelinkMiFlasher.html:814 2021-02-12 14:52:45 # Vbat: 2597 mV , Temp: 1.52°C, Humi: 40.57%, Count: 42
TelinkMiFlasher.html:814 2021-02-12 14:42:41 # Vbat: 2595 mV , Temp: 1.45°C, Humi: 40.64%, Count: 43
TelinkMiFlasher.html:814 2021-02-12 14:32:37 # Vbat: 2595 mV , Temp: 1.43°C, Humi: 39.92%, Count: 44
TelinkMiFlasher.html:814 2021-02-12 14:22:33 # Vbat: 2598 mV , Temp: 1.63°C, Humi: 40.69%, Count: 45
TelinkMiFlasher.html:814 2021-02-12 14:12:29 # Vbat: 2600 mV , Temp: 1.89°C, Humi: 41.43%, Count: 46
TelinkMiFlasher.html:814 2021-02-12 14:02:25 # Vbat: 2598 mV , Temp: 1.88°C, Humi: 41.36%, Count: 47
TelinkMiFlasher.html:814 2021-02-12 13:52:22 # Vbat: 2599 mV , Temp: 1.82°C, Humi: 39.65%, Count: 48
TelinkMiFlasher.html:814 2021-02-12 13:42:18 # Vbat: 2604 mV , Temp: 2.22°C, Humi: 40.15%, Count: 49
TelinkMiFlasher.html:814 2021-02-12 13:32:14 # Vbat: 2604 mV , Temp: 2.3°C, Humi: 40.94%, Count: 50
TelinkMiFlasher.html:810 Memo end: 0
TelinkMiFlasher.html:810 Vbat: 2533 mV , Temp: -4.13°C, Humi: 56.2%, Count: 54, flg: 22
TelinkMiFlasher.html:810 Vbat: 2539 mV , Temp: -4.14°C, Humi: 56.11%, Count: 55, flg: 22
```
</details>

pvvx, 2021-02-13:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/logger.c#L160
```
	mblk.temp = summ_data.temp/summ_data.count;
	int16_t = int32_t  / uint32_t
	--------------- disasm --------------------
    1ac4:	ec29      	tadds	r1, r5, #0
    1ac6:	97fe 9c89 	tjl	3dc <__udivsi3>
    1aca:	20b0      	tstorerh	r0, [r6, #4]
``` 
8-(
The compiler puts:
Runtime Function: unsigned int **__udivsi3** (unsigned int a, unsigned int b)
The function return the quotient of the unsigned division of a and b.

I'll explain to the compiler tomorrow what the function is required: int **__ divsi3** (int a, int b)

skaldo, 2021-02-13:
Wow that was fast, I was even not aware of the averaging functionality 🙂
I just updated my device and it works as expected. Thanks a lot!
[image]

closed 2021-02-13 by skaldo