pvvx/ATC_MiThermometer #45: Documentation: Major feature and improvments of this firmware version

bentolor, 2021-02-13:
Awesome! Thanks for you contributions, @pvvx

I'm a previous user from the original @atc1441 Firmware version https://github.com/atc1441/ATC_MiThermometer and I wonder about the improvements and features of this firmware?

I'd love to some introductory features/highlights bullet point list in the top of the README. Is there a starter so I could propose a MR?

pvvx, 2021-02-13:
Can you explain what needs to be done? I have not mastered github and do not speak English.

bentolor, 2021-02-14:
@pvvx I understand.

I made a proposal at https://github.com/pvvx/ATC_MiThermometer/pull/49

You can preview my changes here: https://github.com/bentolor/xiaomi-mijia-bluetooth-firmware/tree/doc/readme-redsign

I tried to understand as much as possible. 

If you have more info to add (like features) you can write them in russian here and I'll try to translate and add them.

pvvx, 2021-02-15:
Not so simple.

Original Xiaomi firmware has the following features:

Service UUID: ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 

| Characteristic | Description               | Format                                   | Data                                                         |
| -------------- | ------------------------- | ---------------------------------------- | ------------------------------------------------------------ |
| ebe0ccb7       | Time                      | uint32                                   | time in ?                                                    |
| ebe0ccb9       | Data Count                | uint32, uint32                           | index read, index new                                        |
| ebe0ccba       | Index                     | uint32                                   | start read index (for Data Notify)                           |
| ebe0ccbb       | Data Read                 | uint32, uint32, int16, int8, int16, int8 | index, time, temp hi (x0,1), humi hi, temp lo(x0,1), humi lo |
| ebe0ccbc       | Data Notify               | uint32, uint32, int16, int8, int16, int8 | index, time, temp hi (x0,1), humi hi, temp lo(x0,1), humi lo |
| ebe0ccbe       | Temperature Unit          | int8                                     | ?                                                            |
| ebe0ccc1       | Temperature and Humidity  | int16, int8, unt16                       | temp (x0.01), humi, vbat (mV)                                |
| ebe0ccc4       | Batt                      | int8                                     | always = 100%                                                |
| ebe0ccc8       | disconnect                | ?                                        | ?                                                            |
| ebe0ccd1       | clear data                | ?                                        | ?                                                            |
| ebe0ccd7       | comfortable temp and humi | int16, int16, int8, int8                 | temp lo (x0.01), temp hi (x0.01), humi lo, humi hi           |
| ebe0ccd8       | set conn interval         | ?                                        | ?                                                            |
| ebe0ccd9       | para_value_get            | ?                                        | ?                                                            |

Key features:
History is being recorded.
There is a comfort level setting.

pvvx, 2021-02-15:
**Optional clause:**
2. Ensure you enabled "experimental web platform features". Therefore copy the according link (i.e. chrome://flags/#enable-experimental-web-platform-features for Chrome), open a new browser tab, paste the copied URL. Now sten the Experimental Web Platform features flag to Enabled. Then restart the browser.

Works without it.

bentolor, 2021-02-17:
Dear @pvvx !

Thanks for you feedback. 

Unfortunately I missed, that you already copied my changes did additonal work. Why no merge? 

I'm going to close this issue now. Will open another merge request for some fixes.

closed 2021-02-17 by bentolor