pvvx/ATC_MiThermometer #46: bad readings with tasmota32 with advertising type all

simonatackatbris, 2021-02-14:
I have upgraded my  LYWSD03MMC sensors (there are 10 of them) to your firmware (originally either the Xiaomi or atc1441 firmware) and now Im getting a really wierd mis reading when using tasmota32 v9.2.0 using internal BLE.

I get a duplicate sensor with very wrong temperature & dewpoint readings eg

Temp 3021.6 °C
Humidity 40.0 %
Dew point 1680.2 °C
Battery 12 %
clearly wrong

And the mac address is backwards eg it reverses the 6 blocks so instead of left to right they are right to left. (pictures are attached)
[image]

If you use all or custom format.

From the readme.md it appears that atc1441 and custom format advertise the same, but it would appear some of the fields are transmitted differently compared to each other

I didnt know which part the issue is with but as it didnt happen with the other firmware my thinking is it could be this firmware)

rjblake, 2021-02-14:
From Tasmota advice, you can only use either ATC1441 or Custom. Do not use ALL. Also, the ATC1441 and Custom are different formats.

pvvx, 2021-02-15:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Connect and set:
[image]

closed 2021-02-15 by pvvx