pvvx/ATC_MiThermometer #47: [Feature Request]: some way to identify device at power on

simonatackatbris, 2021-02-14:
When you have lots of the same devices it very hard to differentiate between them.  The old atc1441 firmware displayed the last 3 blocks of the MAC address on startup and this was very helpful.  

Is there some way this feature could be reimplimented? Or something to provide similar ability to identify units

pvvx, 2021-02-14:
nRF Connect shows the full MAC and RSSI level. If you bring your smartphone or smartphone close to the device, RSSI level will be lower. This way you can differentiate any device.

rjblake, 2021-02-14:
I printed small labels with the MAC address on them and stuck them on the lower side so can always be read. I have 8x already installed and since they don't move around, not really a big issue

pvvx, 2021-02-15:
The MAC can be changed - re-stick the stickers?

closed 2022-05-01 by pvvx