pvvx/ATC_MiThermometer #48: OTA flashing is not possible with Telink Flasher 1.9

ralf-e, 2021-02-14:
In the current Telink Flasher (1.9), I am currently no longer able to flash the firmware. Does it no longer work with the current version, or am I doing something wrong here?

[image]

PeteBa, 2021-02-14:
I had a similar issue trying to flash an "ATC firmware" device. Flashing a PVVC firmware device worked fine.

To upgrade the ATC firmware device, I ended up using the ATC flasher to get the latest pvvx firmware uploaded and then using the pvvx flasher to set the config parameters. A bit convoluted but it seemed to work.

pvvx, 2021-02-15:
After connecting, all the necessary menus will appear and the correct firmware options will be loaded. 
Otherwise, they constantly confuse and program in LYWSD03MMC firmware from MHO-C401, etc.

PeteBa, 2021-02-15:
That is not what I am seeing when trying to re-flash "ATC firmware" devices.

The pvvx flasher hangs after detecting ATC custom firmware. The flasher will show the "Get Config" button but not "Download Firmware". I waited for over 6 mins. I suspect that config options/format may have changed since ATC and that the detection code is in a wait loop.

See screenshot below:

[image]

pvvx, 2021-02-15:
[image]

PeteBa, 2021-02-15:
Thanks, yes, the issue is when trying to re-flash older ATC firmware with your newer firmware.

As a suggestion, when your flasher detects an unrecognised firmware then it could just show the "download firmware" section but hide the "configure options" items. This would allow the user to upgrade toy your version from an older/unsupported firmware

pvvx, 2021-02-15:
> As a suggestion, when your flasher detects an unrecognised firmware then it could just show the "download firmware" section but hide the "configure options" items. This would allow the user to upgrade toy your version from an older/unsupported firmware

Then many will be flashed into the firmware for another device and there will be many repetitive questions - "does not work ..."

PeteBa, 2021-02-15:
It is a good question and I can see the issue. I notice that the devices look physically different and you may be able to use that. Could you introduce a selection dialog to guide the user in choosing the right firmware ? This would only be needed when the firmware cannot be recognised. Perhaps something like below:

[image]

@ralf-e , I hope I didn't hijack your issue and that this is similar to your circumstances ?

ralf-e, 2021-02-15:
I have now tested the flasher version 1.9 again with some of my sensors.

Telink Flasher for Mi Thermostat 1.9 only offers me a file selection / firmware update if a pvvx-ATC version is already flashed on the respective sensor (MHO or LYWSD) (version 2.0).

This selection is not available for sensors with older ATC versions and also for sensors with original firmware, i.e. a firmware update or flashing of new sensors does not work!

ralf-e, 2021-02-15:
OK, for the sensors with org. Firmware I found my errors now:
Before the update, the sensor must be activated ... (easy to read over) ...

pvvx, 2021-02-15:
This translates into a separate project. I'm not an HTML designer :)
What I can, I will build in ...

pvvx, 2021-02-19:
https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html

closed 2022-05-01 by pvvx