pvvx/ATC_MiThermometer #50: Question: Power consumption vs. Original Xiaomi firmware (LYWSD03MMC)

rjblake, 2021-02-14:
Victor, trying to understand all the graphs and power consumption. If I use the default settings of your firmware, am I right in thinking this use more power than the original Xiaomi formware even when connecting to the device every 5 minutes (for example). Any ideas on the battery life expected using the original firmware on the LYWSD03MMC?

pvvx, 2021-02-15:
Expected battery life when using original firmware on YWSD03MMC, connecting to device every 5 minutes (for example) is expected to be 3 months.

pvvx, 2021-02-15:
In connection mode, everything depends on the initiator. If the external software is capable of working with stand-alone BLE devices, it will change the connection settings to optimal.
By default, the device automatically suggests the optimal settings. But not all external software can handle this. The entire internet is littered with bad examples.

pvvx, 2021-02-15:
Compare:
https://pvvx.github.io/ATC_MiThermometer/OriginalPower.html

https://pvvx.github.io/ATC_MiThermometer/CustPower.html

rjblake, 2021-02-16:
Victor - many thanks for the very clear explanation and excellent details. Really helpful information.

closed 2021-02-16 by rjblake