pvvx/ATC_MiThermometer #52: external sensor?

Ysbrand, 2021-02-17:
Just wondering, have you looked at the possibility of using an external sensor? I'm thinking about monitoring a fridge having the display on the outside, while sensing the inside temperature.

pvvx, 2021-02-18:
Without disassembling the device, only one pin is accessible.

Pin marked "reset":
The I / O state is broadcast in a beacon in a "custom" format.
A pin marked "reset" can be used:
1. as an input.
2. as an output for switching external load depending on temperature and humidity threshold values.

For the rest of the options, the repository contains the source codes of the program.

Ysbrand, 2021-02-18:
Thanks for your answer. I was actually thinking along the line of moving the internal sensor to the outside, but apparently that is not feasible.

I will look for other options.

closed 2021-02-18 by Ysbrand