pvvx/ATC_MiThermometer #53: Not connected for configuration

G1K, 2021-02-18:
Latest firmware 2.3
What can I do?
```

19:29:24: Connecting to: ATC_24CECE
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Reconnect 1 from 5
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Reconnect 2 from 5
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Reconnect 3 from 5
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Reconnect 4 from 5
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Reconnect 5 from 5
19:29:57: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
19:29:57: Something went wrong, to many reconnect's
```

pvvx, 2021-02-19:
Delete all device binding entries on the system.
We connect to the device in Android or Windows and update to the new version...

G1K, 2021-02-19:
[image]
I can't do it

brew-your-own, 2021-02-19:
I get the same thing with the TelinkOTA.html 
I am doing this from linux / chromium (raspbian), how do you remove the device binding entries in the system?
I looked in bluetoothctl but not obvious what to look for

G1K, 2021-02-19:
I flashed the latest firmware ATC_Thermometer23p.bin via USB> UART CP2102, and the web interface is working, you can configure

https://github.com/pvvx/TlsrComSwireWriter

@pvvx I used the address of the beginning 0x00200000, in the example in the repository from 0. How to do it correctly?
It gave an error several times at startup

```
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/ttyUSB0, 921600 baud...
Reset module (RTS low)...
Activate (1000 ms)...
UART-SWS 92160 baud. SW-CLK ~24.0 MHz(?)
Inputfile: ATC_Thermometer23p.bin
Write Flash data 0x00200000 to 0x0020f438...
Error Read Flash Status! (0)   
Error write sector at 0x200100!

```

pvvx, 2021-02-20:
> I used the address of the beginning 0x00200000, in the example in the repository from 0. How to do it correctly?
It gave an error several times at startup

OTA is written from 0x00000 or from 0x20000. Depending on where the current firmware is located.
At startup, the current firmware is checked with 0x00000, and then adding the address ...
If you write the firmware to 0x00000, then the launch will be guaranteed.

pvvx, 2021-02-20:
> It gave an error several times at startup

The SWire timing diagram must be consistent with the UART speed. This is difficult to do and desynchronization occurs. When working with SWire USB-UART, the chip receives a byte frame synchronization error, but in the Chinese versions of the chips it still gives this byte. Reading SWire is based on this. On the FTDI chip, an error syncing bytes does not output bytes to the output and does not work.

brew-your-own, 2021-02-27:
I tried v2.3 of the flasher at https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
but I am unable to connect to the device where I installed the 2.3 firmware:
8:39:49 PM: Connecting to: ATC_20BD2C
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Reconnect 1 from 5
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Reconnect 2 from 5
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Reconnect 3 from 5
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Reconnect 4 from 5
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Reconnect 5 from 5
8:39:55 PM: NotFoundError: No Characteristics matching UUID 00010203-0405-0607-0809-0a0b0c0d2b12 found in Service with UUID 00010203-0405-0607-0809-0a0b0c0d1912.
8:39:55 PM: Something went wrong, to many reconnect's

Is there any way I can flash OTA that particular device with firmware 2.3?

closed 2021-04-15 by pvvx