pvvx/ATC_MiThermometer #55: [Feature Request] Device Name in BLE advertisement

mvadu, 2021-02-21:
With this fork of ATC firmware we can change the name of the device. Now for external tracking it would be great if we can have the BLE advertisement packet contain a device name as well, so we can easily identify the data. (instead of external mac->name lookup).

pvvx, 2021-02-21:
The BLE device scan scheme includes a request for the device name.
Connection to the device is not required and does not cause an increase in power consumption.
[image]
(https://github.com/pvvx/UBIA/tree/master/Sniffer)

pvvx, 2021-02-22:
_**Scanning**_
**Scanning Basics**
When not connected, Bluetooth devices can either advertise their presence by transmitting advertisement packets or scan for nearby devices that are advertising. This process of scanning for devices is called device discovery. There are two types of scanning; active and passive. The difference is that an active scanner can send a scan request to request additional information from the advertiser, while a passive scanner can only receive data from advertising device. Note that the terms discovery and scanning may be used interchangeably. The figure below shows the sequence where the scanner sends a scan request to an advertiser during an advertising event.

[image]

When it comes to the timing of a scan, there are a few parameters you need to get familiar with. Each of these parameters has a range dictated by the [Bluetooth Core Spec](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=286439).

https://dev.ti.com/tirex/content/simplelink_academy_cc2640r2sdk_4_30_01_00/modules/blestack/ble_scan_adv_basic/ble_scan_adv_basic.html

mvadu, 2021-02-22:
@pvvx thank you for that information. It makes sense.

closed 2021-02-22 by mvadu