pvvx/ATC_MiThermometer #56: Clock information not reliable

ercoleg, 2021-02-24:
Hello,
first of all I wish to thank you for what you are doing on this project.
I noticed that the clock information is not reliable, time calculated by the device runs faster than conventional time.
One week ago I set the clock
`11:32:42: Device Date: 2021-02-17 11:32:38.000`
and today I checked the clock
`10:09:23: Device Date: 2021-02-24 10:17:17.000`
The device clock is about 8 minutes in advance, more da 1 minute/day.
Is the issue related to a not reliable internal clock frequency of the device?
can we foresee a parameter to fine tune the alghorithm you are using to calculate the time?

Thank you
Gabriele

pvvx, 2021-02-24:
> I noticed that the clock information is not reliable, time calculated by the device runs faster than conventional time.

The SDK uses binary libraries associated with restoring the time counter after sleep mode.

ercoleg, 2021-02-24:
Just a further question, this means that the reliability of the Time depends on the sleep mode duration?
to you think it is possible to handle a SW correction over the hw counter readings?

Sorry is the question sounds stupid, I'm just a mechanical engineer with a very poor HW background.

pvvx, 2021-02-24:
RC oscillator is unstable. Its course is highly temperature dependent. Calibrating the built-in RC oscillator takes time and increases power consumption. Moreover, calibration must be done often, and this is unprofitable.

The device is cheap and has no quartz for watches.
The clock can be adjusted every day.

ercoleg, 2021-02-24:
Thank you for the explanation, I fully agree.
I have to understand how to automatically adjust time once a day and it will work fine.

pvvx, 2021-02-25:
In version 2.4:
[image]

11:58:28: Send delta time ok
11:58:29: Device StepTimeSec: 1000000.813 us

ercoleg, 2021-02-27:
I have tuned the parameter and now it works perfectly! Thank you!
How does it work the modification you have done? I would like to understand just the principle.

GE

pvvx, 2021-03-13:
Please write what kind of correction "clock delta" you got.

ercoleg, 2021-03-13:
I have set:  Device StepTimeSec: 1000845.125 us
Since I have uploaded new sw 2.6 it started to be a little bit in advance, I have to recalibrate.

G

hpirila, 2021-09-03:
It would be nice also for [Esphome](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi#lywsd03mmc) to synchronize the Xiaomi clock every 6 hours or so.

mateuszdrab, 2021-10-30:
> It would be nice also for [Esphome](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi#lywsd03mmc) to synchronize the Xiaomi clock every 6 hours or so.

Or to have a python script automatically connect to all ATC sensors and re-sync their time, could run this as cron.

Protoncek, 2022-02-12:
I have 3 esp32 modules as BLE receivers for my 8 xiaomi’s. Now i’m testing esp32 with tasmota FW and first results are better than with esphome - in tasmota all seen xiaomi’s are shown seconds after power-on, while in esphome it can take up to a minute. I have some minor issues with wifi disconnecting, but thatks another story and can be solved. And here i can immediately see which modules are in a range of esp module.
So, i can confirm that tasmota is better than esphome in this case, eventhough i’m a bigger fan of esphome, because i can program module exactly the way i want, while tasmota is preprogrammed.

benep, 2022-02-17:
> I'd love to see an automatic clock calibration function added to the website where the last calibration time (stored in the device?) is compared to the current time and the clock delta is calculated based on the drift.

I use 7 devices with your software and I love it. Thank you! All the clocks on my devices running different. So I'd also like an automatic calculation button for the delta in TelelinkMiFlasher.

Maybe it is possible to store the last update datetime of the clock on the device. Then the calculation with the tree times (update, global and device) and the actual delta is easy. If this is not possible you can maybe store the update datetime as cookie on the browser with the mac of the device. That'd only work if you always use the same browser, but it's way better than manual calculation.

pvvx, 2022-02-18:
Receiving and transmitting data in javascript has a jitter of 2.5 seconds (BLE connection interval).
Requires a large counter of seconds since the clock was last set.
The minimum period for the next delta calculation is more than 3 hours.
[image]
Included in version 3.6b.

migreg, 2022-11-28:
> > It would be nice also for [Esphome](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi#lywsd03mmc) to synchronize the Xiaomi clock every 6 hours or so.
> 
> Or to have a python script automatically connect to all ATC sensors and re-sync their time, could run this as cron.

I'm using the delta time option but if someone has a python script for time syncing via cron, please share it. Thanks

jd1, 2023-11-02:
You can also synchronize the time via ESPHome: https://esphome.io/components/display/pvvx_mithermometer.html?highlight=lywsd03mmc#only-synchronize-the-time-once-a-day

mateuszdrab, 2023-11-02:
> You can also synchronize the time via ESPHome: https://esphome.io/components/display/pvvx_mithermometer.html?highlight=lywsd03mmc#only-synchronize-the-time-once-a-day

This actually doesn't work for me, I had it set up like that but the time still drifted away really bad.

akoivist, 2024-09-01:
Wrote (using ChatGPT to be honest) a little python app to set the clock without ESPHome. I scheduled it to be run once a week on a desktop pc.
https://gist.github.com/akoivist/61420a0d22276dadfc4de79fb7e4c776

pvvx, 2026-01-21:
> What do you think @pvvx ?

I think what you wrote is complete nonsense.

> I know I can calculate the drift with the TelinkFlasher site, but wouldn't it be more feasible to apply this into the firmware? 

TelinkMiFlasher calculates the deviation based on the computer's clock.
The hygrometer doesn't know its timekeeping offset.

> I.e. on every (nth) timeset, let's adjust the drift.

The displacement deviation is transmitted to the hygrometer.
The hygrometer's clock is then adjusted every second.
When moving to the next second of counting, a correction is added or subtracted. The correction uses a resolution of 1/16 nanosecond.

danergo, 2026-01-21:
> TelinkMiFlasher calculates the deviation based on the computer's clock.

Yes, I _know_.

But the device itself can also do this calculation _after_ the PC sends the current (correct) date&time to it:

For example in my case:

```
Last clock setting: 2024-12-11 08:28:03
Device local date: 2026-01-22 01:54:25
Current date received via BT: 2026-01-21 20:36:10
```

Firmware now has all information to calculate the deviation. I _know_ this will not be super-precise - but we can calculate the deviation on every timesync, and save it only once per month or so (to not stress the flash too much).

(I know the BT communication can cause delays of seconds, but for a hygrometer device we shouldn't expect Swiss-made watch-precision anyway IMHO)

danergo, 2026-01-21:
> Since Linux has not supported new standards for any hardware for over a decade, the hygrometer was updated with a feature that replaces the BLE PAwR functionality, but is implemented within the legacy BT4.2 standard.
> 
> The interval at which the device sends a beacon with the current timestamp is set. It then waits 7 ms for a response beacon from the device with the specified MAC address with a clarifying timestamp. To account for reception dropouts, the device produces three such beacons. This requires significantly less power than servicing the start of a BLE connection.
> 
> This feature is present in the LYWSD03MMC firmware, but is disabled by default by setting the interval to zero.

PAwR is extremely useful for this! But yes, in general nothing seems to fully supporting it except a few silicons.

It's absolutely phenomenal the firmware is capable for this feature :) You made great progress with these small devices!

> The RC generator's drift is greater than you might imagine and fluctuates depending on a multitude of factors. So what's the point of burdening the device's battery with the additional calculations of these fluctuations?

Honestly, I don't know about the magnitude of the RC generator's drift, and this was just an idea without deep-diving into details. 

I just realized how much drift gets collected within 1 year without setting the clock - but in case I'll set the clock every day, drift will not be too disturbing, or will it?

pvvx, 2026-01-21:
And I no longer see the point in solving the problems of dying Linux in the firmware of BLE devices.

> I just realized how much drift gets collected within 1 year without setting the clock - but in case I'll set the clock every day, drift will not be too disturbing, or will it?

To display the clock without displaying seconds, it's enough to adjust it every couple of days or weeks.
It all depends on the temperature deltas and the device's polling during active scanning. The time count is coarsened and accumulates different deltas depending on the SoC's wake-up time from sleep mode.