pvvx/ATC_MiThermometer #57: Trying to `Select an advertising device` in Advertising.html gives TypeError

mvadu, 2021-02-25:


pvvx, 2021-02-25:
https://stackoverflow.com/questions/64654250/device-watchadvertisements-is-not-a-function-web-bluetoothchrome

mvadu, 2021-02-25:
@pvvx that link does not tell me anything I can check as a consumer, right? I tried with two smartphones, and a laptop to get same behavior. Could you please try once from your end and confirm if its still working?

pvvx, 2021-02-26:
[image]

**Additional tests:**
https://googlechrome.github.io/samples/web-bluetooth/
Watch Advertisements (Promises) / Watch Advertisements (Async Await) - watch advertisements from a BLE device.

**Implementation Status:**
https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#gatt-communication-api

mvadu, 2021-02-26:
Thank you.. I was thinking this page is where I can select the advertisement type? In your code you have old atc version and your own version. Where do I select the type of advertisement sent from thermometer

mvadu, 2021-02-27:
I found the advertising type dropdown in main flasher page itself.

closed 2021-02-27 by mvadu