pvvx/ATC_MiThermometer #58: Incoherent battery readings (also not limited to 100%)

znanev, 2021-02-26:
One of my MHO-C401 devices started to give strange readings for the battery from time to time. As I log those readings in Home Assistant, you can see the strange fluctuations on this graph:

[image]

It appears that sometimes the battery values jump to over 200% (up to 211% in fact). The temperature / humidty measuers at those same times appear perfectly normal, so the strange fluctuations affect the battery measurements only.

I'm not sure why this can be, as the settings for all my MHO-C401 devices are the same, running the same version of the firmware (currently v2.3p). Following is the battery graph of the other MHO-C401 sensor, covering the same time interval:

[image]

As you can see, battery values appear perfectly normal.

pvvx, 2021-02-26:
The current version of the device software has 4 types of battery percentage transmission packets. Your Home Assistant build probably doesn't know all versions and is confusing something.

znanev, 2021-02-26:
I'm only using the unencrypted Mijia format (3 in your list above). Battery measurements are coming as value of general attribute 0x100a. So the erroneous values are definitely coming from the device and are in no way linked to Home Assistant.

I'll try to capture some of the incoherent frames.

znanev, 2021-02-26:
Just checked the advertisement packets at the moment when Home Assistant was showing the unusually high values of 211% - 213%. Indeed no incoherent battery readings are coming in the frames, received from the device.

So the bug is probably a parsing error in [Passive BLE Monitor integration](https://github.com/custom-components/ble_monitor) component. I'll check there. 

Sorry for the false alarm, closing this issue.

closed 2021-02-26 by znanev

pvvx, 2021-02-27:
> I'm only using the unencrypted Mijia format (3 in your list above).

https://github.com/custom-components/ble_monitor recommends a different type of Advertising format.