pvvx/ATC_MiThermometer #59: ESPHome ловит адверты очень редко

vevsvevs, 2021-02-28:
Период адверта установлен раз в 5 секунд (пробовал и другие), но успешно ловятся они совершенно бессистемно - от 20-30 секунд до 20 минут и более. При этом в nRF Connect вижу пакеты с довольно редкими выпаданиями, т.е. вполне стабильно. Полагаю, что связано с параметрами `interval` и `window` на стороне ESP32, но перебрав великое множество разных вариантов, сколько-нибудь надёжного поведения так и не добился. Нативные датчики разных моделей ловятся нормально, сборка от atc1441 - тоже без проблем, всё совпадает с отчётами nRF Connect. Очень бы хотелось понять причину, и использовать эту прошивку совместно с ESPHome.

pvvx, 2021-03-02:
https://esp8266.ru/forum/threads/ochen-chasto-esp32-propuskaet-priem-reklamnyx-paketov.5533/

vevsvevs, 2021-03-02:
Спасибо за ответ. Но с окном 125мс, пропуски от стока (возможно, благодаря "пучковым" отправкам), и от ATC (не знаю благодаря чему) идут примерно в отношении 1/100. Здесь же получается 95/100, и это лучший результат. В любом случае, спасибо за попытку разобраться, будем копать дальше, а пока "боевые" просто верну на ATC.

pvvx, 2021-03-02:
Test reception of beacons (Advertising) [RTL8272DM Realtek AmebaD Arduino](https://github.com/ambiot/ambd_arduino).
Original Xiaomi firmware:
       delta: 16xx ms, lost: 3, total receive: 100
Custom firmware version 2.4:
       delta: 25xx ms, lost: 0, total receive: 150
       ....
       delta: 2524 ms, lost: 1, total receive: 260

pvvx, 2021-03-02:
> Но с окном 125мс, пропуски от стока (возможно, благодаря "пучковым" отправкам), и от ATC (не знаю благодаря чему) идут примерно в отношении 1/100. Здесь же получается 95/100, и это лучший результат. 

ESP32-WROWER-B, Arduino, sketch: 
https://github.com/pvvx/ATC_MiThermometer/blob/master/esp32/BLE_term_lcd/BLE_term_lcd.ino

---

Firmware: 
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer.bin 19 Nov 2020
Time scanning 517 seconds, lost: 36, rx total: 224:
```
Time since boot: 517. New scan.
  ------ delta: 1900 ms, lost: 36, total: 224
Found service '181a' data len: 17, 10161a18a4c1380b5eed01121f630c1b15
MAC: a4c1380b5eed
Temp: 27.4°, Humidity: 31%, Vbatt: 3099, Battery: 99%, count: 21
```

---

Firmware v2.4:
https://github.com/pvvx/ATC_MiThermometer/blob/master/ATC_Thermometer24.bin
Time scanning 731 seconds, lost: 41, rx total: 236:
```
Time since boot: 731. New scan.
  ------ delta: 2500 ms, lost: 41, rx total: 236
Found service '181a' data len: 19, 12161a18ed5e0b38c1a4b90a4a0cf20b5f6004
MAC: a4c1380b5eed
Temp: 27.45°, Humidity: 31.46%, Vbatt: 3058, Battery: 95%, flg: 0x04, count: 96
```
Где  95/100 ?

pvvx, 2021-03-03:
Оставил в людном месте в городе на 8 часов. От приемников до датчика около 10 метров.
Кол-во сканируемых окружающих устройств BLE колеблется от 50 до 100, кол-во WiFi станций - аналогично.

Последние сообщения в логах:
ESP32: delta: 2520 ms, lost: 1941, rx total: 9884
RTL8722DM: delta: 2510 ms, lost: 853, rx total: 11601

vevsvevs, 2021-03-03:
Тоже проверил оба варианта (ATC и PVVX) в нагруженной среде, результаты плачевны, очевидно есть какие-то проблемы в реализации ESPHome. Судя по всему, лучшие результаты ATC были "ошибкой выжившего", т.к. я проверял её только по адвертам с данными, т.е. со значительно бОльшим интервалом выборки.. Предвижу ответ, но всё же спрошу: может всё-таки сочтёте возможным реализовать burst-отправку адвертов с отдельным от маяка интервалом? Ведь проект ESPHome - очень популярное, гибкое и удобное решение во множестве кейсов в рамках HASS, и очень жаль что его сканер BLE подводит в столь типовом сценарии использования.

pvvx, 2021-03-03:
Ещё раз проверил что там показывает тест.
ESP32 перезагрузился - можно считать вышел из строя, хотя у него подключено резервирование питания на АКБ.
RTL8722DM продолжает считать:
delta: 2497 ms, lost: 1764, rx total: 23530
Т.е. в самой неблагоприятной обстановке потери рекламных пакетов от датчика составляют 7.5%.
Это при шуме в эфире от других BLE приблизительно в 3000 пакетов в сек и от 50-ти работающих WiFi AP.
Магазин у метро, Cанкт-Петербург.

pvvx, 2021-03-03:
> Ведь проект ESPHome - очень популярное, гибкое и удобное решение во множестве кейсов в рамках HASS, и очень жаль что его сканер BLE подводит в столь типовом сценарии использования.

Я не могу переписать все сторонние программы и исправить все кривые разработки чипов от Espressif и других производителей не вписывающихся в стандарты BLE.

pvvx, 2021-03-03:
К примеру [sniffer на TLSR8266 чипе](https://github.com/pvvx/UBIA/tree/master/Sniffer) ловит практически все пакеты от всех моих домашних BLE устройств. 
У ESP32-WROWER-B (у меня есть и другие ESP32 – различий не заметил), в документации  значится:
[image]

А у  TLSR8266: 
[image]
По указанным значениям он хуже, но в модуле JDY-10 на внутреннюю антенну ловит лучше.
Кому после этого верить?

vevsvevs, 2021-03-03:
Безусловно, в первую очередь надо верить личному опыту :) Я с вами полностью согласен, виновный определен, а в последнем сообщении я лишь просил по возможности сделать костыль для одного из самых распространенных и доступных DIY-проектов, т.к. на исполнительном устройстве это выглядит реальнее, чем на изначально забагованном железе ESP32. Ответ я получил, ещё раз спасибо за исчерпывающие ответы и отличный проект!

closed 2021-03-03 by vevsvevs