pvvx/ATC_MiThermometer #60: -5% humidity with custom firmware in comparison with OFW normal?

shyney7, 2021-02-28:
Hi,
after flashing my devices with the cfw 2.4 they all have a -5% difference in the readings of the humidity. The temperature seems to be the same as on the stock firmware.
For example if one device with OFW reads 58% humidity the device with CFW reads 53%.

I know that I can fix this by adding a offset of 5% in the config. But I just want to know if this is on purpose? And if so why?

pvvx, 2021-03-01:
The original has an offset inserted. So like Xiaomi.

shyney7, 2021-03-01:
> The original has an offset inserted. So like Xiaomi.

So is the original value with build in offset more accurate than the cfw non offset value or vice versa?

pvvx, 2021-03-01:
See sensor documentation.
https://www.mouser.com/datasheet/2/682/Sensirion_Humidity_Sensors_SHTC3_Datasheet-1386761.pdf
Custom firmware returns values directly from the sensor.

pvvx, 2021-03-01:
The difference in the readings of different sensors can be found here
https://wiki.liutyi.info/display/ARDUINO/Test+0+v9+balcony+high+humidity

shyney7, 2021-03-01:
> 
> 
> See sensor documentation.
> https://www.mouser.com/datasheet/2/682/Sensirion_Humidity_Sensors_SHTC3_Datasheet-1386761.pdf
> Custom firmware returns values directly from the sensor.

Thank you! So if I understand it correctly the sensors are calibrated by the sensor manufacturer fulfilling ISO standards.
>The calibration  of  the  reference,  used  for  the  calibration  of  the transfer  standards,  is  NIST  traceable  through anISO/IEC 17025 accredited laboratory.

So the raw readings should be more accurate then the ones from the OFW with offset. Altough if I compare it with my DHT22 sensors in the same locations running on D1 minis or NodeMCU with ESPHome the readings from the mijia sensors seems to be the same using the offset in comparison with the ESPHome readings without offset.
[image]

If I remove the offset from the config I would get not only lower readings in comparison with the stock firmware but also with the readings from my DHT22 / AM2302 sensors...

closed 2021-03-01 by shyney7

rjblake, 2021-03-01:
> > The difference in the readings of different sensors can be found here
> > https://wiki.liutyi.info/display/ARDUINO/Test+0+v9+balcony+high+humidity
> 
> Sorry for the above reply I didnt see your post about the sensor comparison. This is actually a very usefull comparison. Thank you!
> It also seems to correspond to my measurements and the comparison with the AM2302 sensor values. The AM2302 readings are 4-5 % higher than the raw readings from the SHTC3 sensor running on cfw. Since the OFW has the exact same offset build in maybe the chinese developers used the AM2302 sensor to "calibrate" the offset 😄

You may want to take a look [here](https://github.com/JsBergbau/MiTemperature2) [scroll down to 'Calibration section'] regards sensor calibration to validate & adjust the actual accuracy of the sensors.