pvvx/ATC_MiThermometer #64: Xiaomi Home + ESP32 readings

DavidAntonin, 2021-03-03:
Hello,
is it possible to have both type of communication? I would like to have sensors connected to Xiaomi Home application but also be able to read temp/hum from ESP32 in arduino project.
Thank you

pvvx, 2021-03-04:
Mi-Home uses the registration of BLE sensors via Bluetooth in the smartphone and each time checks the registration keys, and also reads the values ​​from the device memory using its own protocol. This is done when connecting to the device from a smartphone. 
Sample: 
https://github.com/pvvx/ATC_MiThermometer/issues/54
The original Xiaomi LYWSD03MMC sends out promotional packages with sensor data with a period of 10 minutes or more. The Xiaomi server logs data through the gateway in 1 hour increments.
The publication of Xiaomi certificates (keys) for registering devices is a violation of the user agreement when using Mi-Home. For this reason, I cannot support Mi-Home.

DavidAntonin, 2021-03-04:
I understand communication between LYWSD03MMC and Mi Home. But with original firmware there is still advertisement I can receive with ESP32:
Found service 'fe95' data len: 26, 191695fe58585b058edf26b038c1a43590ae051900009ab8800b
MAC: a4c138b026df
Crypted data[11]! count: 142
Found service 'fe95' data len: 26, 191695fe58585b058edf26b038c1a43590ae051900009ab8800b
MAC: a4c138b026df
Crypted data[11]! count: 142
Found service 'fe95' data len: 26, 191695fe58585b058edf26b038c1a43590ae051900009ab8800b
MAC: a4c138b026df
Crypted data[11]! count: 142
Found service 'fe95' data len: 16, 0f1695fe30585b058fdf26b038c1a408
MAC: a4c138b026df

I have a Mi Token and Mi Bind Key, can I use it to decode this data?

I Also found a bug when I revert back original firmware - Hardware version is cleared (0000)  Before it was B1.4

pvvx, 2021-03-04:
If you register a device with the original firmware, then after restoring the bind key for the beacon in the custom firmware ("Show all mi keys") and enabling the encryption of the beacon for the "Encrypted Mi beacon" type, the Xiaomi gateway will continue to receive values from the sensor and publish them in the cloud ... But working from a smartphone in Mi-Home with a sensor will be impossible.
If ESPHome can work with bindkey, it will receive data from the sensor.

closed 2022-05-01 by pvvx