pvvx/ATC_MiThermometer #66: Need help with Advertising interval

chris4git, 2021-03-07:
Hi,
thanks to share your work.
May you help to understand the calculation of the advertising interval ?

If I set
_Advertising interval 5000 , step 62,5 ms_
would the interval be 5000 X 62,5 ms = 312,5 s
or 5000 s modulo 62,5 = 5000 s ?
or 5000 ms modulo 62,5 = 80 s ?

In fact, with this value, I see an advertisement every 20040 ms !

Confusing !

pvvx, 2021-03-10:
[image]

chris4git, 2021-03-10:
Ok, it's clear. So I don't understand why I receive broadcasts only once every ~20000 ms when I put 5000 as advertising interval :-/

pvvx, 2021-03-13:
> So I don't understand why I receive broadcasts only once every ~20000 ms when I put 5000 as advertising interval :-/

Where, in which packet receiving sniffer, do you receive broadcasts in 20 second increments when the ad interval is 5 seconds?
The change of data in advertising occurs after a set number of advertising intervals.

chris4git, 2021-03-14:
> Where, in which packet receiving sniffer, do you receive broadcasts in 20 second increments when the ad interval is 5 seconds?
The change of data in advertising occurs after a set number of advertising intervals.

It's my own application, built on top of Linux Bluez stack 5.43.
Example of traces :
```
387C51 seen 19792 ms ago (1615717646)
387C51 seen 19841 ms ago (1615717665)
387C51 seen 19821 ms ago (1615717685)
387C51 seen 19772 ms ago (1615717705)
387C51 seen 39592 ms ago (1615717745)
387C51 seen 19812 ms ago (1615717764)
387C51 seen 19821 ms ago (1615717784)
```
When I use Android BLE Scanner it gives advertising ranges from 5000 to more than 10000 ms but often around 5000.
So I suspect my application or Bluez...

pvvx, 2021-03-16:
[Testing for several days on RTL8722DM in a noisy radio environment](https://github.com/pvvx/ATC_MiThermometer/issues/59#issuecomment-789889723):
Packets Lost: 1764, Total Received: 23530 Packets
Advertising interval 2.5 seconds.

pvvx, 2021-03-16:
[Sniffer](https://github.com/pvvx/UBIA/tree/master/Sniffer) next to Xiaomi LYWSD03MMC:
[image]

jenshenk, 2022-01-17:
II don't know whether the OP found a way to fix it but just by accident I fixed it in my case and wanted to share my experience since I thought maybe it helps towards perfecting this nice project!
I've a couple of thermostats flashed with the atc1441 firmware and wanted to give this fork a try since I liked all the additional options, especially the clock. Just for context: I use BLE-tasmota flashed ESP32s with the "MI32Option2 1" to act as a BLE to MQTT bridge and forward all broadcasts immediately.
Initially I had a similar experience like the OP: All atc1441 flashed thermostats were regularly broadcasting as configured only the pvvx one was quite inconsistent ranging from every 10s to sometimes 2 minutes. I thought maybe it's the clock function, so there's no temperature broadcast during the clock is shown yet switching the clock off didn't change anything. However, switching it on again led to a constant broadcast like the atc1441 ones. Very happy with it since now I've a clock in my bathroom in addition to broadcasts that work as expected.

closed 2022-05-01 by pvvx