pvvx/ATC_MiThermometer #67: I have one Thermometer in a weird state

JustDevZero, 2021-03-08:
I tried last firmware and somehow the termometter is in weird state.

Telink flasher tells me the following:

1:26:14: Searching for devices
1:26:20: Connecting to: LYWSD03MMC
1:26:28: Detected custom Firmware
1:26:28: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html

But If i try to see through Mi Home is still with 81.4 official.

I tried removing from Mi Home an adding again, but still the same.

Any ideas?

Thanks

pvvx, 2021-03-08:
The official firmware uses the BLE name "LYWSD03MMC".
After flashing the custom firmware, you should have a device named "ATC_xxxx".
If the firmware is not complete, then the new device will not appear.

pvvx, 2021-03-08:
> But If i try to see through Mi Home is still with 81.4 official.

"B1.4" - is the hardware version number.

JustDevZero, 2021-03-09:
Yeah, wrote 81.4 but it's B1.4.

If i try to connect though TellinkMiFlasher now shows the following:

2:08:03: Searching for devices
2:08:07: Connecting to: LYWSD03MMC
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Reconnect 1 from 5
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Reconnect 2 from 5
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Reconnect 3 from 5
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Reconnect 4 from 5
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Reconnect 5 from 5
2:08:14: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
2:08:14: Something went wrong, to many reconnect's

Version hardware: B1.4
Firmware: 1.0.0_0109

I think something it's... fucked up :(

Weird thing its that Mi Home can identify it, and if I try through [MiTemperature2](https://github.com/JsBergbau/MiTemperature2) script i can get the data:

./LYWSD03MMC.py  -d A4:C1:38:XX:XX:XX -r -b
Temperature: 19.4
Humidity: 76
Battery voltage: 2.97 V
Battery level: 87

rjblake, 2021-04-06:
@JustDevZero - not sure on the flashing, but I noticed on battery that my thermometers went from 100% down to 88% in the course of 7 days, but have only dropped another 1% over the next 1.5 months. Could be the nature of the CR2032 batteries.

MacSass, 2021-04-06:
> 
> 
> @JustDevZero - not sure on the flashing, but I noticed on battery that my thermometers went from 100% down to 88% in the course of 7 days, but have only dropped another 1% over the next 1.5 months. Could be the nature of the CR2032 batteries.

Uh, this is interesting - because I have seen the same with my LYWSD03MMC (Custom FW 1.7 still):

It was running at 84% battery for about 2 weeks - then suddenly it fell down to 70% from one day to the next (it stayed in the same location in my office - no change in environment). No connected device - only using advertising frams with 1.7 FW default settings.
From 70% it started crawling up again - about 1% every day up to 82%. Now it has been at 81% for about a week with no change. Somehow the battery measurement seems to have some funny flaws ...

atc1441, 2021-04-06:
The battery meassurement uses 3.1V as 100% but the battery is in reality 3V 100% this is done to have a better overview of how full the battery really is with just the % value

MacSass, 2021-04-06:
Thanks, that should explain why we would never see 100% - but I was more interested to understand why it fell by 14% in on day and then started to creep up again ... nearly to the previous level.

MacSass, 2021-04-06:
In my case it only dropped by 14% about 2 weeks after the OTA update - which I found strange.
But anyhow - I would not consider it so important at all ...

atc1441, 2021-04-06:
Yes, if the battery is empty the screen will be off, very simple^^ 

And if no direct connection is possible anymore the battery is low

pvvx, 2021-04-07:
It may not be a battery drain, but a poor battery contact. This is observed on several thermometers.
Battery current and voltage are not sufficient to break through the oxide layer (potential barrier).

pvvx, 2021-04-07:
> The battery meassurement uses 3.1V as 100% but the battery is in reality 3V 100% this is done to have a better overview of how full the battery really is with just the % value

Some CR2032 batteries have an initial voltage of 3.3..3.4 V.
Because Li battery, it is possible to charge it. And the charge even of the old one is kept above 3.3V.

daniel88ferreira, 2021-10-03:
Does someone has an idea on what can be done to fix this?

I can confirm that now, through TellinkMiFlasher, I'm getting "No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device."

But, I can still pair the with Mi Home and I can also get the temp and humidity values using, for example, [this]( http://www.d0wn.com/using-bash-and-gatttool-to-get-readings-from-xiaomi-mijia-lywsd03mmc-temperature-humidity-sensor/). So the device is advertising unencrypted values. Also the device name is still LYWSD03MMC not ATC_xxxx.

Thanks in advance!

closed 2022-05-01 by pvvx