pvvx/ATC_MiThermometer #68: What determines when the device sends an update?

penright, 2021-03-08:
I felt like I was highjacking https://github.com/pvvx/ATC_MiThermometer/issues/23 question. An @MichielioZ pointed out that I was getting confused between the device sending advertising and updates. I just assumed that the device would send data with every advertising packet. I can see a power-saving by not turning on the measuring sensor, but if you are turning on the radio to send advertising would sending whatever the last data use that much more power? 
What is the maximum time between updates?

**_Let me give a little more context for my question._** 
I have just started with ATC device and ESPHome. Seeing the update was asynchronous and ESPHome would not know if a device quit reporting, I started trying to figure out how to know if a device has not sent an update in X time. That way HA could send an alert that temp sensor XYZ is down. I was going to use the state property "last_updated".  Because of some HA and ESPHome internals, I was going to create a Heartbeat based on the "on_value" event in ESPHome. That is when I notice that there are long times between data updates. That is why I am asking for the maximum time between updates, so I know when to send an alert that a sensor may be down.

pvvx, 2021-03-08:
Device send data with every advertising packet.
The period for the transmission of advertising packages is set as:
[image]
The time between data updates in advertising packages is equal to the sensor polling time:
[image]
In advertising packages, a counter for polling sensors is transmitted to detect data updates.

ESPHome has flaws in the ESP32 BLE drivers. The scan settings prevent ESP32 from receiving packets all the time and there are many unaccepted packets. Other equipment is free of these shortcomings.

apaperclip, 2021-03-11:
@pvvx 
Does this help with the scanning flaw you mention?

esp32_ble_tracker:
  scan_parameters:
    active: false

pvvx, 2021-03-13:
> Does this help with the scanning flaw you mention?
> 
> esp32_ble_tracker:
> scan_parameters:
> active: false

These are bugs in the ESP32 BLE driver SDK.
Other SoCs work well - no more than 8% dropouts in heavily clogged radio air.
https://github.com/pvvx/ATC_MiThermometer/issues/59

penright, 2021-03-16:
@pvvx I hope my questions did not come across as complaining. I was just trying to understand. As I mention in "setting context", I connected the dots that whatever the receiver is would not know when a device was offline other than X amount of time with no updates. So I started different tests seeing what should I set X time for. The first lesson learned is HA only updates when a value changes. So even when the device and ESP are communicating, the last_updated value was not changing. So I created a "heartbeat" sensor that the ESP creates and updates on each received from the device. That is when I asked about what are the expectations. From what I have read here and saw in practice, my X time, with default settings, is around 15 minutes. In fact, my ESP could not connect to HA, which a reboot of HA is what fixed it, help me see it in this graph. You can see the stretch in dark blue when it was the same value for a long time. Right now the time is from the ESP and I am not worried about it being right or better said "set right", just that it is changing.
[image]

closed 2022-05-01 by pvvx

pvvx, 2022-11-22:
After the "Read memory measures" command, the thermometer changes the "connection interval" to a [faster one](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/cmd_parser.c#L485). 
After sending the requested number of measurements, the "Connection latency" is restored.
Otherwise, reading 19632 measurements takes a very long time.

[Default "Connection interval"](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/ble.h#L42) is 20ms. And "Connection latency" is designed for 2.5 seconds.
As a result, in normal connection mode, packets follow on average every 2.5 seconds, and during "memory measures" transmission - 20 ms.

swiergot, 2022-11-22:
Ok, so I tried to build the firmware with `bls_pm_setManualLatency(24)` in line 485, which should be 500 ms. But the transfer of records takes the same time as before.

I changed `cmd_parser.c` and ran `make`, then took `ATC_Thermometer.bin`. What am I missing?

pvvx, 2022-11-22:
"Connection delay" does not affect the transmission interval from the device. "Connection delay" is the amount of time a device may not respond or accept requests. But it can transmit with the "Connection interval" step. The receiver still needs to be prepared to accept every "connection interval". 
See the BLE standards for a more precise description. 
I do not know English well and it is difficult for me to describe through Google translator...

pvvx, 2022-11-22:
In the "connection intervals" you can change the ESP32 by sending a request after connecting.

swiergot, 2022-11-22:
I can't see any option to change it after connecting:

https://docs.micropython.org/en/latest/library/bluetooth.html

Maybe MicroPython does not expose this API.

swiergot, 2022-11-22:
Based on the links you provided, it seems that even setting a longer connection interval does not guarantee that packets will be transmitted slower, as multiple packets can be transmitted in a single connection interval.

pvvx, 2022-11-23:
No more than the set MTU can be transmitted in one interval.
The size of one package depends on the version of BT. 
Default maximum MTU size of 23 bytes -> one packet to support BT4.0.
As a result, 1 packet is transmitted during the "Connection interval" time.

---

Arduno ESP32 Test...

I inserted a piece of code 
```
		uint8_t blk[3];
        blk[0] = 0x35;
        blk[1] = 50;
        blk[2] = 0;
        Serial.printf("*** Get Memo\r\n");
        pRemoteCharacteristic->writeValue(blk,3); // get Memo
```
 into the [sketch for ESP32](https://github.com/pvvx/ATC_MiThermometer/blob/master/esp32/BLE_term_lcd/BLE_term_lcd.ino) for testing. 
And I get the following logs via UART:
```
Found service '181a' data len: 19, 12161a18faf80538c1a4fefea122dc0a400a0f
MAC: a4c13805f8fa
Temp: -2.58°, Humidity: 88.65%, Vbatt: 2780, Battery: 64%, flg: 0x0f, count: 10
*** Get Memo
Time since boot: 246. New scan.
 ------ delta: 5006 ms, lost: 38, rx total: 44
Found service '181a' data len: 19, 12161a18faf80538c1a4fefea122dc0a400a0f
MAC: a4c13805f8fa
Temp: -2.58°, Humidity: 88.65%, Vbatt: 2780, Battery: 64%, flg: 0x0f, count: 10
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3501004ae97d63bb09f00d140c
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3502003de67d63a1090b0e1f0c
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35030031e37d637e092a0e870c
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3504003ee07d636a09630e890c
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3505005add7d6321099a0e290c
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3506006eda7d6361093f0e330a
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3507003bea7d63f008370f3909
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35080052e77d6306099b0f3409
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35090082e47d630009b80f3709
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350a00b2e17d63fa08be0f3409
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350b00e2de7d63f308bc0f3109
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350c0012dc7d63eb08940f3009
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350d0042d97d63ef086d0f3109
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350e0072d67d63f0089d0f3109
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 350f00a2d37d63ea088d0f3009
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 351000d1d07d63e0085f0f2f09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35110001ce7d63f008d20f3109
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35120031cb7d63ec0807103009
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35130061c87d63e30818102f09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 35140091c57d63d50825102e09
[... 42 lines cut ...]
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352a00af877d632b08bb101209
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352b00de847d631908bd101009
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352c000e827d630008b6100e09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352d003e7f7d63ed0792100c09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352e006e7c7d63e40783100c09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 352f009e797d63ea078a100c09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 353000ce767d63ed078a100c09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 353100fe737d63ec075c100d09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 13
data: 3532002e717d63ee074d100e09
Notify callback for characteristic 00001f1f-0000-1000-8000-00805f9b34fb of data length 3
data: 350000
```

50 measurements are transferred to the ESP32 in 12 seconds
[image]
(power СR2032  mA/ms)

pvvx, 2022-11-23:
ESP32 cannot successfully handle BLE in Arduino at 160MHz CLK. Even at 240 MHz, packet acknowledgments are delayed, which greatly affects the batteries of the sensors.

swiergot, 2022-11-23:
Excellent point. I don't intend to query the sensor during a normal operation, but only when if ESP32 for some reason has gaps in readings from advertisements. Besides, I suppose the library can be easily changed to have fixed service indices.

Big thanks for all your help!