pvvx/ATC_MiThermometer #69: Can't set the RF tx power

oangelo, 2021-03-09:
I am tring to set the tx power on Telink Flasher, but, when I press the button "Send Config" the minimum power is automaticaly set. "Set default" and  then "Send Config" works to go back to the default. I wish to improve the tx power to be able to receive the data a bit more distant from my bluetooth hub. This hapens both in my smartphone and on a linux browser.

pvvx, 2021-03-10:
?
[image]

oangelo, 2021-03-10:
Today it is working, thanks @pvvx. But, I can not choose the power higher than +3.01 dbm, only lower. Higher power options were showing yesterday. Is 3.01dbm the maximum power?

pvvx, 2021-03-11:
The microcircuit has 2 antenna options (VBAT and VANT) and an RF power supply. The second option works with reduced power on the current board ...

ibamsey, 2021-03-22:
So what is the highest power setting that can be achieved on the LYWSD03MMC ?

pvvx, 2021-03-22:
> So what is the highest power setting that can be achieved on the LYWSD03MMC ?

Depends on the internal resistance of the battery.
The chip has tx output power up to +10 dB

pvvx, 2021-12-12:
Replacing the CR2032 with a more powerful battery allows for higher TX RF power to +10 dB. A pulse current of 30 mA is required at 3.3 V.
In TelinkMiFlasher.html, [change line](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L28): 
```` const MAX_RF_TX_Power = false; ````
to 
```` const MAX_RF_TX_Power = true; ````
[image]

closed 2023-02-18 by pvvx