pvvx/ATC_MiThermometer #70: Receive data from elsewhere?

wigster, 2021-03-10:
Hi,

Would it be possible to have the sensor actually receive some data from elsewhere (e.g. my ESP32) and display it? I am thinking that rather than displaying the temperature it is measuring, it would be fun if it could display some other temperature, e.g. an average from the house than I calculate from all the sensors, or cycle between internal and external temperature that I send to it.

Is something like this possible, or does the BLE interface not work at all like that?

Thanks!

pvvx, 2021-03-11:
https://github.com/pvvx/ATC_MiThermometer/tree/master/esp32/BLE_term_lcd

vekonyat, 2021-03-15:
Hi, with Victor's firmware it's also possible using a Linux machine with Gatttool, I'm using an RPI4 for that, running a script from a Home Assistant. (Well on Hassio I couldn't do it, because Gatttool was not implemented, so I had to SSH from HA into the Pi and run the Python script there:...) Plus with a little modification in the firmware you can change it to show the external data only.

pvvx, 2021-03-16:
>  Plus with a little modification in the firmware you can change it to show the external data only.

The firmware has two methods of output to the LCD/E-ink.

1. Reception formatted data transmission, with interleaved display.
2. Working with raw LCD buffer. Bitwise control of indicator segments. Permanent display when connected.

closed 2021-03-29 by pvvx