pvvx/ATC_MiThermometer #71: With BLE fitness trackers ?

mr-sneezy, 2021-03-10:
I'll post this as an idea rather than a feature request, as it's probably not possible or in the scope of the original environmental sensor integration concept but I won't know unless I ask...

Does anyone other than me wonder if the same concept used with MiThermometer could be used with cheap hackable BLE fitness wrist bands that have body temp, heart rate, blood oxygen monitoring etc ? 

I.E. to reflash them with branched firmware that will in similar fashion advertise body data to HA via the ESPhome BLE hub without any OEM apps or cloud based monitoring (privacy intrusion). 
The benefits of this would be detailed health monitoring at home (particularly during sleep) and HA automations based on the sensors, with warnings/logs about high heart rate, low blood oxygen etc etc. 

Thoughts anyone ?

atc1441, 2021-03-10:
That is possible the same way yes

Actually i got 20 of some 2€ fitness tracker with telink soc just never started with them as no one could benefit from it, when they are hard to get it doesnt make much sense.
If there is a cheap device widely available it would be perfect

pvvx, 2021-03-10:
[image]
[image]
....

mr-sneezy, 2021-03-11:
Interesting. Clearly you've given the idea some thought. 
I have a couple of China fit bands here too, neither of which I use because I found the OEM apps and firmware too clunky and lacking in some way, like proper at home logging.
Happy to buy another to hack/test if one of the many models is suitable. It does not need much of a display, just enough to see if it's working.

pvvx, 2021-03-12:
> I.E. to reflash them with branched firmware that will in similar fashion advertise body data to HA via the ESPhome BLE hub without any OEM apps or cloud based monitoring (privacy intrusion).

Unfortunately, ESPHome (ESP32) is very bad at receiving beacons (Advertising Packages).

mr-sneezy, 2021-03-13:
Ah OK. 
Hey, but isn't that what we are picking up from the reflashed MiThermometers sensors now though ?
(very sorry if I've found a gap in my understanding of the advertising meaning, I know it's not your job to educate me) 

Reason I ask is that ESPHome Hub is what I'm using for the 4x sensors spread all over my house (including one that is inside the fridge) with reading reliability.

closed 2021-04-15 by pvvx