pvvx/ATC_MiThermometer #73: cannot add back to mihome after flashing

xjustbmw, 2021-03-10:
is it possible to have the sensor work again also in mihome ?
i first flashed the initial firmware (by atc1441) and then your version
Im getting couldn't verify in mihome app (-28) error

If i click 'show all my keys'  i get no midevid, no mitoken and nomibindkey!

PS. I made a screenshot of the midevid mitoken and mibind key while flashing first atc firmware

pvvx, 2021-03-11:
The atc1441 firmware removes all keys.
Mi-Home has additional registration keys. The Mi-Home user agreement prohibits third party use and any reverse-engineering.

xjustbmw, 2021-03-11:
cool. not that im really interested. but in this case, after restoring to back to original firmware , it still works with mihome ?

anybody using an old android phone as a daemon for these sensors ?

pvvx, 2021-03-11:
If the registration was made on the original firmware and the keys restored in custom-fw, plus "Advertising type: mi" and "Encrypted Mi beacon - enable" - Xiaomi Gateway 3 works, but unstable.

Xiaomi Cloud stores one measurement point per hour.

xjustbmw, 2021-03-11:
is there any way to manually restore the keys ? i have them (as picture) but they're not on the device  :)

pvvx, 2021-03-11:
Binding and re-registering will not take place without the original firmware.

atc1441, 2021-03-11:
You can delete the thermometer in MiHome and re add it with the stock firmware

serhat-ozkara, 2021-03-11:
Subscribing, exactly what i'm going through. What's the update interval on stock firmware? It's lagging a lot on stock firmware with gateway 3. That's why i'm looking for a custom firmware..

pvvx, 2021-03-12:
> What's the update interval on stock firmware?

About 10 minutes.

 > It's lagging a lot on stock firmware with gateway 3.

Updating data in the cloud through a gateway - 1 hour.
Updating data on a smartphone when it is connected to the sensor. Each connection of the smartphone to the sensor checks the registration of the sensor and sends the data to the cloud.

pvvx, 2021-03-12:
Registration procedures and examples of algorithms from mijia are in mi_api.
There are even headlines in Tuya:
https://github.com/TuyaInc/tuya_smesh_sdk_tlsr825x_light
"tuya_smesh_sdk_tlsr825x_light \ includes \ board \ chip \ telink_sig_mesh_sdk \ sdk \ vendor \ common \ mi_api"
The mi_api procedure code can be found on the Telink website and other mijia members.
The encryption library itself from mijia is private.
Registration requires encryption keys associated with each type of mijia device:
certificate[>400 bytes] и dev_cert_pri[32 bytes]

The Mi-Home user license prohibits any reverse engineering. Copying keys and using the cloud is prohibited.

serhat-ozkara, 2021-03-12:
@pvvx I think there's an issue with send encrypted option. If i select mi advertisement along with encrypted checked, neither mi app nor passive ble integration on hass can see the device. I have actual registered beaconkey and secret installed on device.

But, i found a really good workaround while tinkering:
If I use original Device name "LYWSD03MMC" AND mi advertising without encryption, device gets connected to gateway3 and gateway3 starts reporting really really quickly on HASS. Passive BLE integration also works but i don't prefer it because Rpi bluetooth is unstable.
Some debug info on the logs:
Process BLE Fix {'did': 'blt.1.160qml0ft4g00', 'eid': 4109, 'edata': 'e6005301', 'pdid': 1371, 'seq': 237}
2021-03-12 13:23:39  DEBUG    gateway3      192.168.155.100 | MQTT | ble/blt.1.160qml0ft4g00 b'{"temperature":23.0,"humidity":33.9}'

xjustbmw, 2021-03-12:
mijiatemp apk seems to be stable enough
since i have a phone lieing around doing other things,i'll parse the notification's temperature/humidity through tasker every x minutes

pvvx, 2021-03-12:
> @pvvx I think there's an issue with send encrypted option. If i select mi advertisement along with encrypted checked, neither mi app nor passive ble integration on hass can see the device. I have actual registered beaconkey and secret installed on device.

If you register a device with official firmware in mi-home, and then flash an alternative firmware and restore bindkey, then gateway 3 sees and logs the device in the cloud.

pvvx, 2021-03-12:
> mijiatemp apk seems to be stable enough
> since i have a phone lieing around doing other things,i'll parse the notification's temperature/humidity through tasker every x minutes

mijiatemp uses the connection and shows open UUIDs. Xiaomi LYWSD03MMC is not secure from connection, but registration is limited.

xjustbmw, 2021-03-25:
is there a maximum working limit for the advertising interval ?
can i set it to something like 180000ms ?
still possible to connect to the sensor after that and change the settings ?

thanks

pvvx, 2021-03-26:
> is there a maximum working limit for the advertising interval ?
> can i set it to something like 180000ms ?

[image]

closed 2022-05-01 by pvvx