pvvx/ATC_MiThermometer #75: How to stop ExtShow?

ChooTingFeng, 2021-03-17:
First of all, thanks for this custom firmware, it is so powerful and there's so many settings

Just 1 question, I hit the ExtShow thing that makes it show 1234, a weird face and 99.

And it has been cycling through that, normal temps and humidity, and clock

How do I stop it from showing that 1234 weird face?

Set Default config didn't seem to reset it

closed 2021-03-17 by ChooTingFeng

OmgImAlexis, 2021-06-03:
Same issue here. I can't find anything to revert this.

PRSXFENG, 2021-06-03:
Hey there, I figured out you just have to wait, one of the variables that
was there is time, so just let it sit for 10 minutes or so and it'll be
back to normal again.

OmgImAlexis, 2021-06-03:
I ended up reflashing. Good to know though.