pvvx/ATC_MiThermometer #77: Add support for BLE long range mode

jan-be, 2021-03-18:
Based on [this datasheet](http://wiki.telink-semi.cn/doc/ds/DS_TLSR8251-E_Datasheet%20for%20Telink%20BLE+IEEE802.15.4%20Multi-Standard%20Wireless%20SoC%20TLSR8251.pdf) the SoC supports 125kbps BLE long range mode (S8) and 500kbps BLE long range mode (S2).

This could help increase the range at the cost of shorter battery life.
The downside is that I think most receivers don't support it because it is an optional feature of Bluetooth 5.0 .

Maybe consider adding optional support for this feature.

pvvx, 2021-03-19:
The chip also supports ZigBee.
All this requires additional non-standard equipment.

closed 2021-03-19 by pvvx

BarryHuffman, 2021-06-13:
> All this requires additional non-standard equipment.

Do you have any more information on this?
Does it involve PCB modification or just something like a antenna?

jan-be, 2021-06-13:
I assume he means non-standard equipment on the receiving end.
Most Bluetooth devices indeed don't support it, however there are some exceptions like many newer Android smartphones and [at least this one usb receiver](https://www.m2mgermany.de/shop/bl654-usb-bluetooth-v5-adapter.html), which is a bit more expensive and most notably the new [ESP32-C3](https://www.espressif.com/en/news/ESP32_C3).

BarryHuffman, 2021-06-14:
Thanks for the clarification @jan-be