pvvx/ATC_MiThermometer #78: about encrypted beacon

50494554524F, 2021-03-22:
can you tell me how should it work?! where do i set the key?!

it not clear on the flasher page...

thanks

pvvx, 2021-03-24:
[image]

pvvx, 2021-03-24:
Version 2.8:  Added saving bindkey to EEP if mi-keys are erased

closed 2021-03-24 by pvvx

50494554524F, 2021-03-24:
i had atc1441 before your firmware... now i have another problem...

updated to 2.8, ok. set a pincode, ok. tried to connect to the flasher, it asks the pincode, connected, ok. 

after this first connection it doesn't connect anymore, remains stuck on "connect to: ....." ...

pvvx, 2021-03-24:
Check the battery. Re-flashing strongly drains the battery.

50494554524F, 2021-03-24:
i spent a bit of time trying... this is the way to reproduce the problem

- windows 10, chrome or edge browser
- firmware 2.8 on a brand new LYWSD03MMC
- set pincode, i used six digits pincode
- disconnect
- reconnect, insert the pincode, ok
- disconnect again
- it won't connect anymore
- remove the device from bluetooth devices remembered by windows
- reconnect, the pincode will be asked again and the connection will be activated
- disconnect
- it won't connect anymore
- again, we have to remove the device from the list....

pvvx, 2021-03-24:
Windows 10 supports only one bind device.

50494554524F, 2021-03-24:
sorry, i don't understand... i'm using only one thermometer at the moment...
the thermometer remains in use by windows after chrome stops using it, right?! ok, this isn't  a problem, can be managed onece in a while...
i'm just adding the pincode to prevent people near me can "hijack" the thermometer...

now, i have to figure out how the bindkey works because home assistant/ble_monitor says always unavailable...

pvvx, 2021-03-24:
**Windows 10 and BLE bind + pin-code:**

The first connection and entering the pin code. 
The device appears in the BT list: 
[image]
The following connections will be successful as long as no other BLE devices with pin-code and binding are connected.

We take out the USB-BT adapter and change it to another one.
Everything - the bind device becomes inaccessible!
Entering a pin-code will not help:
[image]
Only removing from the BT list and a new bind and entering a pin code will help!

Likewise, if you already have another binding device in the BT list. Windows 10 will not connect and will ask for a pin code endlessly, ending with an error message.

Contact to Microsoft :)

50494554524F, 2021-03-24:
oh ok, so no... i'm using only this thermometer at the moment... windows or chrome don't ask for the pincode or connect anymore... btw not a big problem.

pvvx, 2021-03-24:
> sorry, i don't understand... i'm using only one thermometer at the moment...
> the thermometer remains in use by windows after chrome stops using it, right?! ok, this isn't a problem, can be managed onece in a while...

Windows 10 will sometimes connected  with a thermometer to show the battery level.

> i'm just adding the pincode to prevent people near me can "hijack" the thermometer...

Windows needs to have this only device in the BT list for Chrome to connect. Then Chrome will connect.
 
> now, i have to figure out how the bindkey works because home assistant/ble_monitor says always unavailable when i turn on the ecncryption...

Packages can be viewed by sniffer. The encoding of the encrypted packet is of [the mijia standard.](https://github.com/pvvx/ATC_MiThermometer/tree/master/InfoMijiaBLE). 
[An example of decoding.](https://github.com/pvvx/ATC_MiThermometer/blob/master/test_adv_key.py)

pvvx, 2021-03-24:
> oh ok, so no... i'm using only this thermometer at the moment... after the first successful connection, windows or chrome don't ask for the pincode or connect anymore... btw not a big problem.

BLE pairing in Windows 10 is stored in the driver and adapter.
Windows 10 only supports one pairing on common BT adapters. 👎

50494554524F, 2021-03-24:
omg the battery, it's clear now... 

about the bindkey: started from scratch with original firmware and reinstalled the 2.8... now it's working, i don't know why... ok

thanks for your patience

50494554524F, 2021-03-25:
sorry to disturb you again... yesterday i was able to set and use the bindkey starting from scratch
after few tests i think the encryption doesn't work, i'm using MI format and encrypted beacon... here it is the steps to reproduce the problem

- original firmware
- do activation, copy the bindkey
- use the bindkey, packets are decrypted in homeassistant/ble_monitor and i can read the temperature
- do activation, copy the new bindkey
- flash firmware v2.8
- read mi keys, the bindkey is what we copied and expected
- select mi packets format, check encrypted beacon, send config
- use the new bindkey as done before ----> NOT WORKING

thanks

pvvx, 2021-03-26:
If you have not erased mi keys, then Mi bindkey has priority.

50494554524F, 2021-03-26:
more confusing now :D... i was supposed to use the bindkey returned by "do activation", right?!

50494554524F, 2021-03-26:
erased mi bindkeys, read eep bindkey and used it... not working
erased mi bindkeys, set my eep bindkey and used it... not working

at this point, mi format and encryption beacon don't work or i don't understand how to use them together... trying other formats

pvvx, 2021-03-26:
erased mi bindkeys, set new bindkey и check:
[image]

50494554524F, 2021-03-26:
i did exactly this... btw homeassistant/ble_monitor works with original firmware and bindkey, i'll ask them to check too

pvvx, 2021-03-26:
Used **mi** bindkeys и check:
[image]

pvvx, 2021-03-26:
> i did exactly this... btw homeassistant/ble_monitor works with original firmware and bindkey, i'll ask them to check too

After removing Mi keys and if something goes wrong, it is advisable to restart the device.

Reboot command: '721234':

[image]

The device will reboot after closing the connection.

50494554524F, 2021-03-27:
someone told me how to enable debug for ble_monitor... i get a tons of "invalid data" rows, these rows disappear when i disable the encryption... so there's a bug somewhere in ble_monitor/firmware/homeassistant, my knowledge finish here and have to put this problem back in the line...

(selecting the custom format permits to use the encryption?!)

pvvx, 2021-03-27:
Ver 2.9 - Added additional id flags to advertising packages
Perhaps this will help if your program does not parse BLE Advertising Packages into blocks.

[0x02,0x01,0x06] is inserted at the beginning of the packet:

https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/

Data Type Value | Data Type Name | Reference for Definition
-- | -- | --
0x01 | «Flags» | Bluetooth Core Specification:Vol. 3, Part C, section 8.1.3 (v2.1 + EDR, 3.0 + HS and 4.0)Vol. 3, Part C, sections 11.1.3 and 18.1 (v4.0)Core Specification Supplement, Part A, section 1.3