pvvx/ATC_MiThermometer #79: Data format mi-like with encryption

Ernst79, 2021-03-27:
I'm the developer of the Passive BLE Monitor integration and we received an issue that his sensor didn't work when he set the sensor to advertisement type `Mi` with encryption enabled. I checked it and can to the same conclusion. 

So I compared a HCI dump message from stock firmware (first line in the table below) to a message from ATC firmware with advertisement type Mi and encryption enabled (second line in the table). As you can see, three bytes are missing (`01 06 1A`). Is it possible to fix/change this. 

```
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HCI  Evt Len Sub Num Evt  Peer -------MAC-------   Len Len Type Val Len  AD  Xiaomi Frame Product Frame     MAC (LE)        ----------------------PAYLOAD------------  RSSI
type code    evt rep type addr                             flag         type  UUID   ctrl  type    cnt                          cypher            ext.cnt     MAC tag
 A    B  C    D  E    F    G          H            I   J    K   L   M    N     O      P     Q      R           S            
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 04   3E 2A  02  01   00   00  C7 A8 D9 38 C1 A4   1E  02   01  06  1A   16  95 FE  58 58  5B 05   90   C7 A8 D9 38 C1 A4   2D E3 2D 7B 49        00 00 00  EF 49 D6 2E  CE   LYWSD03MMC (default firmware)
 04   3E 27  02  01   00   00  B2 18 8D 38 C1 A4   1B  1A                16  95 FE  58 58  5B 05   2C   B2 18 8D 38 C1 A4   A1 BA 47 2B 4C        00 00 00  1A 7D A5 6F  CD   LYWSD03MMC (ATC, mi-like, encrypted)
```

pvvx, 2021-03-27:
Ver 2.9 - Added additional id flags to advertising packages
Perhaps this will help if your program does not parse BLE Advertising Packages into blocks.

[0x02,0x01,0x06] is inserted at the beginning of the packet:

https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/

Data Type Value | Data Type Name | Reference for Definition
-- | -- | --
0x01 | «Flags» | Bluetooth Core Specification:Vol. 3, Part C, section 8.1.3 (v2.1 + EDR, 3.0 + HS and 4.0)Vol. 3, Part C, sections 11.1.3 and 18.1 (v4.0)Core Specification Supplement, Part A, section 1.3

pvvx, 2021-03-27:
[Mijia Telink samples ](https://yadi.sk/d/g5fV7WD1EaUdRQ)
mi_release_version_4_1_1\ble_sdk_multimode\vendor\common\mijia_ble\libs\common\mible_beacon.c

Ernst79, 2021-03-27:
Thanks, I’ll check tomorrow. 

BLE monitor is splitting the message up, but it first does a check on this 02 01 06 part to be present. As it was missing, parsing of the message failed. I could have worked around this, but it seems better to have the exact same format as the original stock firmware. 

Anyway, thanks for the fast response, I’ll test it tomorrow.

pvvx, 2021-03-27:
> I could have worked around this, but it seems better to have the exact same format as the original stock firmware.

The project does not aim to make a fake Xiaomi LYWSD03MMC.

Ernst79, 2021-03-28:
I can confirm it's ok working now. Thanks

> The project does not aim to make a fake Xiaomi LYWSD03MMC.

I do understand. I just noticed a difference in the BLE advertising format, which made BLE monitor not parsing the message (due to the check on 02 01 06 (or 15 16 95) being present in the message). This check works for all other Xiaomi MiBeacon sensors, so my thought was that it would be the easiest to fix in your firmware. 

> These three bytes now increase battery consumption by 5%. This is what the 'users' wanted. :)

Hmmm, that's significant. Only by adding these three bytes? I'll leave it up to you, to decide on whether you keep the change or not. It can also be fixed on the side of BLE monitor, if you decide to reverse.

closed 2021-03-28 by Ernst79

50494554524F, 2021-03-28:
i was testing too... now packets are decrypted but read randomly 5-15 minutes by homeassistant, while i have set the firmware to send the packets every minute

if i can, as it's permitted by the protocol, i would go back to firmware 2.8 and correct ble_monitor component...

thanks both

pvvx, 2021-03-28:
I exaggerated that 5%.
More precisely, then:
the length of the message is about 40 characters + preamble
Increase in TX duration no more than 3/40.
TX has the highest current, which is more than two times than all other operating modes of the chip. This is at +3 dB.
Reducing TX power has a stronger effect on average consumption, levels over 0 dB.
As a result, these 3 bytes affect the level of energy consumption of one percent.

> if i can, as it's permitted by the protocol, i would go back to firmware 2.8 and correct ble_monitor component...

In version 2.8 there are errors in the mode of working with encrypted advertisements in alternation by count.
The change time for encrypted advertisements is 16 measurements from the sensor with default settings. 
For LYWSD03MMC, this is 160 seconds = 16 * 4 ad interval periods.
Temperature, voltage and battery values are rounded off, recorded and broadcast in the next 64 advertisements.

Ernst79, 2021-03-28:
@50494554524F In addition, let me explain how BLE monitor works. These 64 advertisements are all the same, with the same `frame counter` (message number). BLE monitor checks this frame counter and only uses one message. If another message is received with the same frame counter, it will ignore it (as it is the same data). So, update interval will be about 160 seconds in HA (which is already faster than the stock firmware, which has a 10 minute interval. If you want more/faster updates, use the `custom` advertisement type instead.

pvvx, 2021-03-29:
> These 64 advertisements are all the same, with the same `frame counter` (message number). BLE monitor checks this frame counter and only uses one message.
64 Advertisings -> 4 message from different data recurring 16 times. 16 Temperature, 16 humidity, 16 battery + 16 empty message for compatibility with the original. All 64 have the same `Frame Counter` (Message Number).

pvvx, 2021-03-29:
>  So, update interval will be about 160 seconds 

The update interval depends on the settings:
1. Advertising interval
2. Number of Advertising Intervals until the next measurement by the sensor
As a result, the data update time will be equal to 16 measurement intervals from the sensor.

50494554524F, 2021-03-29:
really clear now, thanks