pvvx/ATC_MiThermometer #80: Connetion not possible after flashing the FW to Version 2.9

Flugente-19, 2021-03-29:
Hello pvvx,

today I have flashed the "MHO-C401 Custom Firmware Version 2.9" to my device MHO-C401. The device seems to work properly and I can receive the temperature, the humidity and the battery status with your test program Advertising.ino. Additionaly I'd like to configure the device according to your description, but the connection with the TelinkMiFlasher.html is not possible. On my Windows 10 notebook I have tried to connect to the device with two different browsers (Chrome and Edge) und got the same error in both cases:

[image]

Do you have an idea, how can I solve the issue?

Thank you in advance!

Kind regards
Flugente-19

pvvx, 2021-03-31:
MHO checked. Works on smartphones with Android and on computers with new BT adapters.

Checked more options for adapters:

[image]

Option "1" works well as many others.

Found a variant of bad work with the adapter number "2":

[image]

Internal error occurs: _DOMException: GATT operation not permitted._

Later I will try to solve the problem in Telinkmiflasher.html ...

pvvx, 2021-03-31:
N2 adapter error detected. These are bugs in Windows drivers and adapter. The adapter remembered the MAC from the previously connected bind and pin code and was buggy. Changing the MAC on the device solved the problem.
MHO-С401 firmware and TelinkMiFlasher.html have nothing to do with it.

Flugente-19, 2021-04-01:
Hello Viktor, thank you for the fast feedback.

Today I checked the connection to the MHO-C401 from my Android smartphone - the connection was possible and the FW configuration could be changed.

But I don't understand, why it was possible to connect and flash your FW from my notebook with the build-in BT 4.2 adapter Intel AC 8265 and afterwards the connection with the same notebook was not possible. I will observe the issue.

Kind regards
Michael

closed 2021-04-13 by Flugente-19