pvvx/ATC_MiThermometer #81: ClearGrass Temp & RH problem

AlexHighTower, 2021-04-02:
Hello, 

I have ClearGrass Temp & RH sensor, it looks like CGG1
[Flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) shows it as "ClearGrass Temp & RH" at list when I press "Connect" and after I select it in list and press "Pair" I got following error:

> 15:26:34: Searching for devices
> 15:26:38: Connecting to: ClearGrass Temp & RH
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Reconnect 1 from 5
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Reconnect 2 from 5
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Reconnect 3 from 5
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Reconnect 4 from 5
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Reconnect 5 from 5
> 15:26:39: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:26:39: Somtthing went wrong, to many reconnect's
> 15:26:46: Disconnected.

at the moment "Disconnect" appears display blinks (refresh)

if I press "Connect" again - it tells "already paired" and a little bit different log at the beginning

> 15:27:42: Searching for devices
> 15:27:47: NotFoundError: User cancelled the requestDevice() chooser.
> 15:27:47: Reconnect 1 from 5
> 15:28:01: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:28:01: Reconnect 2 from 5
> 15:28:01: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:28:01: Reconnect 3 from 5
> 15:28:01: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:28:01: Reconnect 4 from 5
> 15:28:01: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:28:01: Reconnect 5 from 5
> 15:28:01: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
> 15:28:01: Something went wrong, to many reconnect's
> 15:28:07: Disconnected.

pvvx, 2021-04-03:
CGG1 variants produced by [Qingping.co](https://www.qingping.co/temp-rh-monitor/overview) ([ClearGrass](https://www.cleargrass.com/))
---

**CGG1-old** ver1,  Confirmed FCC ID: [2AQ3F-CGG1](https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168)
https://pvvx.github.io/CGG1_old
nRF52810 Chipset

**CGG1-?** ver2, Unknown FCC ID.
https://github.com/kelchm/cgg1-thermometer-firmware
nRF52832 Chipset

**CGG1-M** ver3,  has an official fake FCC ID: 2AQ3F-CGG1
https://pvvx.github.io/CGG1
TLSR8253 Chipset

**CGG1-H** ver4, Confirmed FCC ID: [2AQ3F-CGG1H](https://fccid.io/2AQ3F-CGG1H/Internal-Photos/Internal-Photos-4723738)
Unknown Chipset

---

Alternative firmware is only possible for **CGG1-M**.
The first version of **CGG1-old** is in development. Will not be in this repository.

kelchm, 2021-04-03:
> ClearGrass Temp & RH sensor, it looks like CGG1

Unfortunately the ClearGrass name almost certainly means you have the older nRF based hardware, making it incompatible with this project. To confirm, you should check the advertised UUID using something like nRF Connect as @pvvx describes [here](https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812851300).

closed 2021-04-15 by pvvx