pvvx/ATC_MiThermometer #84: gatttool command to read sensor values (temp, humi, battery voltage, date/time)

manu-t, 2021-04-06:
Hi,
I'm starting with BLE sensors and I wonder if it is possible to retreive data with the gatttool command?
I can read temperature and humidity with the characteristics 0x2A6E and 0x2A6F but I don't know how to get the battery voltage and the sensor's internal date/time.
Do I need to write a command to get these values?
Thanks!

pvvx, 2021-04-07:
gatttool does not work with BLE advertising packets.

gatttool is used to access the “services” running on bluetooth device in connection-mode.

https://github.com/pvvx/ATC_MiThermometer#bluetooth-connection-mode

https://github.com/pvvx/ATC_MiThermometer#control-function-id-when-connected

Examples can be found in  https://github.com/pvvx/pvvx.github.io/tree/master/ATC_MiThermometer
GraphAtc2.html, GraphMemo.html, TelinkMiFlasher.html

manu-t, 2021-04-07:
ok, thanks!

closed 2021-04-08 by pvvx